commaai/openpilot #27665: FPv2: consider extra ecus for present ecu brand matching

sshane, 2023-03-24:


sshane, 2023-03-24:
Can make a project if we don't this cluttering up issues

MoreTore, 2023-04-10:
Currently trying to debug a vehicle and I'm wondering if you could take a look? Here is a drive 539cfef88f5a6208|2023-04-05--13-49-04. It's a Mazda CX50 and I can't get it to completely fingerprint. Apparently, their Comma 3 was not shipped with a comma power, however, bus 1 should be connected to the OBD port at the BCM (see this wiring diagram below).

[image]

CAN BUS 0/2 is used for communication between the following modules:
― BCM
― EPS control module
― Adaptive front lighting system (AFS) control module/auto leveling control module
― SAS control module  'address': N/A
― DSC HU/CM
― TCM 
― PCM

'carFw': [{'address': 1840,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'eps',
                          'fwVersion': b'VA40-3216Y-B\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': False,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1848,
                          'subAddress': 0},
                         {'address': 1798,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'VA40-67WK2-\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': False,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1806,
                          'subAddress': 0},
                         {'address': 1892,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'VA45-67XK2-\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': False,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1900,
                          'subAddress': 0},
                         {'address': 1888,
                          'brand': 'ford',
                          'bus': 0,
                          'ecu': 'abs',
                          'fwVersion': b'VA40-4300F-D\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
                          'logging': False,
                          'obdMultiplexing': False,
                          'request': [b'>\x00', b'"\xf1\x88'],
                          'responseAddress': 1896,
                          'subAddress': 0}],

incognitojam, 2023-04-19:
@MoreTore The `carFw` you posted shows responses for `brand: 'ford'`, so it won't fingerprint as a Mazda since there are no responses for Mazda FW queries. You should try adding those queries to Mazda's FP config.

MoreTore, 2023-04-19:
> 1888

Thank you. I didn't understand what was going on there. I will try that.

MoreTore, 2023-04-20:
@incognitojam 

Thanks alot that solved it!

  (0x730, None): [
    b'VA40-3216Y-B\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
  ]
  (0x706, None): [
    b'VA40-67WK2-\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
  ]
  (0x764, None): [
    b'VA45-67XK2-\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
  ]
  (0x760, None): [
    b'VA40-4300F-D\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
  ]

Number of dongle ids checked: 1
Fingerprinted:                0
Not fingerprinted:            1
  of which had a fuzzy match: 0

Correct fuzzy matches:        0
Wrong fuzzy matches:          0

sshane, 2023-04-20:
When ECUs don't respond and you expect them to, you can check `errorLogMessage` in useradmin and see what error is being printing for that ECU. If the rx address (usually + 8) exists in ecu responses, you usually just need a different data id for that car and ECU.

MoreTore, 2023-04-20:
> When ECUs don't respond and you expect them to, you can check `errorLogMessage` in useradmin and see what error is being printed for that ECU. If the rx address (usually + 8) exists in ecu responses, you usually just need a different data id for that car and ECU.

Well yes I see errors but I have very little understanding of the whole of what to do.

```
{'errorLogMessage': '{"msg": "vin query retry (1) ...", "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", "version": "0.9.2", "dirty": '
                    'true, "device": "tici"}, "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": "car_helpers.py", "lineno": 101, '
                    '"pathname": "/data/openpilot/selfdrive/car/car_helpers.py", "module": "car_helpers", "funcName": "fingerprint", "host": "tici", '
                    '"process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": 1680716945.1326694}',
 'logMonoTime': 56865860445,
 'valid': True}

....

{'errorLogMessage': '{"msg": "Error processing UDS response: (1882, None)", "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", '
                    '"version": "0.9.2", "dirty": true, "device": "tici"}, "exc_info": "Traceback (most recent call last):\\n  File '
                    '\\"/data/openpilot/selfdrive/car/isotp_parallel_query.py\\", line 108, in get_data\\n    dat, updated = msg.recv()\\n  File '
                    '\\"/data/openpilot/panda/python/uds.py\\", line 445, in recv\\n    self._isotp_rx_next(msg)\\n  File '
                    '\\"/data/openpilot/panda/python/uds.py\\", line 502, in _isotp_rx_next\\n    assert not self.tx_done, \\"isotp - rx: flow '
                    'control with no active frame\\"\\nAssertionError: isotp - rx: flow control with no active frame", "level": "ERROR", "levelnum": '
                    '40, "name": "swaglog", "filename": "isotp_parallel_query.py", "lineno": 110, "pathname": '
                    '"/data/openpilot/selfdrive/car/isotp_parallel_query.py", "module": "isotp_parallel_query", "funcName": "get_data", "host": '
                    '"tici", "process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": 1680716946.9520495}',
 'logMonoTime': 58696144038,
 'valid': True}
{'errorLogMessage': '{"msg": "Error processing UDS response: (1882, None)", "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", '
                    '"version": "0.9.2", "dirty": true, "device": "tici"}, "exc_info": "Traceback (most recent call last):\\n  File '
                    '\\"/data/openpilot/selfdrive/car/isotp_parallel_query.py\\", line 108, in get_data\\n    dat, updated = msg.recv()\\n  File '
                    '\\"/data/openpilot/panda/python/uds.py\\", line 445, in recv\\n    self._isotp_rx_next(msg)\\n  File '
                    '\\"/data/openpilot/panda/python/uds.py\\", line 502, in _isotp_rx_next\\n    assert not self.tx_done, \\"isotp - rx: flow '
                    'control with no active frame\\"\\nAssertionError: isotp - rx: flow control with no active frame", "level": "ERROR", "levelnum": '
                    '40, "name": "swaglog", "filename": "isotp_parallel_query.py", "lineno": 110, "pathname": '
                    '"/data/openpilot/selfdrive/car/isotp_parallel_query.py", "module": "isotp_parallel_query", "funcName": "get_data", "host": '
                    '"tici", "process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": 1680716947.0009077}',
 'logMonoTime': 58727973257,
 'valid': True}
{'errorLogMessage': '{"msg": "iso-tp query bad response: (2016, None) - 0x7f2231", "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", '
                    '"version": "0.9.2", "dirty": true, "device": "tici"}, "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": '
                    '"fw_versions.py", "lineno": 294, "pathname": "/data/openpilot/selfdrive/car/fw_versions.py", "module": "fw_versions", '
                    '"funcName": "get_fw_versions", "host": "tici", "process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": '
                    '1680716947.0209897}',
 'logMonoTime': 58748993205,
 'valid': True}
{'errorLogMessage': '{"msg": "iso-tp query bad response: (2017, None) - 0x7f2231", "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", '
                    '"version": "0.9.2", "dirty": true, "device": "tici"}, "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": '
                    '"fw_versions.py", "lineno": 294, "pathname": "/data/openpilot/selfdrive/car/fw_versions.py", "module": "fw_versions", '
                    '"funcName": "get_fw_versions", "host": "tici", "process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": '
                    '1680716947.0216691}',
 'logMonoTime': 58749541226,
 'valid': True}
{'errorLogMessage': '{"msg": "iso-tp query bad response: (2016, None) - 0x7f2231", "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", '
                    '"version": "0.9.2", "dirty": true, "device": "tici"}, "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": '
                    '"fw_versions.py", "lineno": 294, "pathname": "/data/openpilot/selfdrive/car/fw_versions.py", "module": "fw_versions", '
                    '"funcName": "get_fw_versions", "host": "tici", "process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": '
[... cut ...]
                    '"vin_rx_addr": 0, "error": true}, "ctx": {"daemon": "controlsd", "dongle_id": "539cfef88f5a6208", "version": "0.9.2", "dirty": '
                    'true, "device": "tici"}, "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": "car_helpers.py", "lineno": 174, '
                    '"pathname": "/data/openpilot/selfdrive/car/car_helpers.py", "module": "car_helpers", "funcName": "fingerprint", "host": "tici", '
                    '"process": 46510, "thread": 547969865552, "threadName": "MainThread", "created": 1680716951.6599796}',
 'logMonoTime': 63388235078,
 'valid': True}
```

adeebshihadeh, 2024-05-17:
@sshane any of this still relevant after all the recent FP work?

sshane, 2024-05-17:
The remaining here are more for my reference for future improvements (not needed now).

closed 2024-05-17 by sshane