commaai/openpilot #27682: ui: crash when scrolling nav destinations

sshane, 2023-03-24:
While onroad, while a destination is set

sshane, 2023-03-24:
It actually hangs, not crash

```bash
[Switching to thread 1 (Thread 0x7fb0c31b70 (LWP 63984))]
#0  __lll_lock_wait (futex=futex@entry=0x16470f8, private=0) at lowlevellock.c:52
52      in lowlevellock.c
(gdb) bt
#0  __lll_lock_wait (futex=futex@entry=0x16470f8, private=0) at lowlevellock.c:52
#1  0x0000007fb7a52cec in __GI___pthread_mutex_lock (mutex=0x16470f8) at pthread_mutex_lock.c:115
#2  0x0000007faf6af880 in EglWaylandWlWindowSurface::DequeueBuffer() () from /lib/aarch64-linux-gnu/libeglSubDriverWayland.so
#3  0x0000007faf6af914 in EglWaylandWlWindowSurface::GetBuffer(EglSubResource*, EglMemoryDesc*) () from /lib/aarch64-linux-gnu/libeglSubDriverWayland.so
#4  0x0000007fb4f52e48 in ?? () from /lib/aarch64-linux-gnu/libGLESv2_adreno.so
#5  0x0000007fb4f54904 in ?? () from /lib/aarch64-linux-gnu/libGLESv2_adreno.so
#6  0x0000007fb4f0b030 in ?? () from /lib/aarch64-linux-gnu/libGLESv2_adreno.so
#7  0x0000007fb4f0a51c in ?? () from /lib/aarch64-linux-gnu/libGLESv2_adreno.so
#8  0x0000007fb4e69d1c in ?? () from /lib/aarch64-linux-gnu/libGLESv2_adreno.so
#9  0x0000007fb7204ca4 in QPlatformBackingStore::composeAndFlush(QWindow*, QRegion const&, QPoint const&, QPlatformTextureList*, bool) ()
   from /lib/aarch64-linux-gnu/libQt5Gui.so.5
#10 0x0000007fb7533430 in ?? () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#11 0x0000007fb75350a4 in ?? () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#12 0x0000007fb7536154 in ?? () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#13 0x0000007fb75362e8 in ?? () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#14 0x0000007fb754eb44 in QWidgetPrivate::syncBackingStore() () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#15 0x0000007fb7566924 in QWidget::event(QEvent*) () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#16 0x0000007fb75244ac in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#17 0x0000007fb752dad8 in QApplication::notify(QObject*, QEvent*) () from /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#18 0x0000007fb6b74c0c in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/aarch64-linux-gnu/libQt5Core.so.5
#19 0x0000007fb6b77b80 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /lib/aarch64-linux-gnu/libQt5Core.so.5
#20 0x0000007fb6bd0d20 in ?? () from /lib/aarch64-linux-gnu/libQt5Core.so.5
#21 0x0000007fb36dd944 in g_main_context_dispatch () from /lib/aarch64-linux-gnu/libglib-2.0.so.0
#22 0x0000007fb36ddbb4 in ?? () from /lib/aarch64-linux-gnu/libglib-2.0.so.0
#23 0x0000007fb36ddc54 in g_main_context_iteration () from /lib/aarch64-linux-gnu/libglib-2.0.so.0
#24 0x0000007fb6bd01cc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/aarch64-linux-gnu/libQt5Core.so.5
#25 0x0000007fb6b733b8 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/aarch64-linux-gnu/libQt5Core.so.5
#26 0x0000007fb6b7ba5c in QCoreApplication::exec() () from /lib/aarch64-linux-gnu/libQt5Core.so.5
#27 0x0000000000423288 in main (argc=<optimized out>, argv=0x7ffffff448) at selfdrive/ui/main.cc:29
```

pd0wm, 2023-03-27:
This is known bug in the EGLWayland driver. You end up with two threads stuck waiting on a condition variable. My guess is this happens when you render too many frames in a short time and some buffer fills up.

closed 2023-06-22 by adeebshihadeh