commaai/openpilot #27695: CANParser: parse all signals from specified messages

adeebshihadeh, 2023-03-25:
Many of the lines in a car port are spent on whitelisting signals to be parsed out. The CANParser should be fast enough now that we can simply whitelist messages and parse out all their signals.

- [ ] remove the signals arg from the CANParser input, see https://github.com/commaai/opendbc/pull/610 a while ago as a starting point
- [ ] add a test case to the [car unit tests](https://github.com/commaai/openpilot/blob/master/selfdrive/car/tests/test_models.py) to ensure parsing time is reasonable
- [ ] if the test fails, optimize the CANParser until it doesn't

adeebshihadeh, 2023-03-25:
@deanlee want to try this one out?

deanlee, 2023-03-26:
Submitted a small pr to improve the performance of query_latest. https://github.com/commaai/opendbc/pull/795

I'm still reading https://github.com/commaai/opendbc/pull/610 to see if there are other parts that need to be optimized.

adeebshihadeh, 2023-03-27:
> I'm still reading [commaai/opendbc#610](https://github.com/commaai/opendbc/pull/610) to see if there are other parts that need to be optimized.

I think we should start with the test case to ensure all the optimization PRs are strictly improving performance. Would be nice to have a simple check on parsing performance in opendbc, in addition to the one for cars that will live in openpilot.

adeebshihadeh, 2023-07-10:
@sshane how's this coming along?

closed 2023-08-11 by sshane