commaai/openpilot #27703: Steering icon ignore double taps

royjr, 2023-03-27:
### Describe the bug

When tapping the steering icon in the upper right corner to switch between chill and experimental modes, a second tap is sometimes registered causing the mode to change twice rapidly which results in the mode basically staying the same.

This is most likely to happen while driving while the device and user are in motion. A small button tap delay such as 0.5s may help with this.

### Provide a route where the issue occurs

1a5d045d2c531a6d

### openpilot version

master

sshane, 2023-04-07:
Not sure exactly how the touch event works, but you might be able to override the event handler and do nothing if a double tap is registered

MoreTore, 2023-04-10:
I was just working on something similar in another project. I will make a pr to fix this.

closed 2023-06-29 by incognitojam