commaai/openpilot #27710: ui: nav does not show when entering a destination after the first time

sshane, 2023-03-28:
nav does not show the second time you set a destination

royjr, 2023-03-28:
Nav: only open map once on destination set https://github.com/commaai/openpilot/pull/21650
https://github.com/commaai/openpilot/blob/3555a08c17af6c1ef22d75345a78ff6f670b1d66/selfdrive/ui/qt/maps/map.cc#L129-L132

sshane, 2023-03-28:
It looks like clearing the nav destination should reset `allow_open`. But since I tapped another destination with a route already set, it didn't pop up again.

sshane, 2023-06-23:
@incognitojam this should be a simple one to knock out

sshane, 2023-06-25:
Also think it should exit the nav settings view when it gets a new route from your phone

incognitojam, 2023-07-11:
I'm not really sure the original issue is a bug - if you hide the map while navigating do you always want it to open again if you change the destination?

adeebshihadeh, 2023-07-11:
Yes, I think so.

sshane, 2023-08-08:
@deanlee want to do this? We should just mirror what navd.py does:

https://github.com/commaai/openpilot/blob/7f02cb9fafb05e77dcbd2b11f41ecb8244f641fe/selfdrive/navd/navd.py#L109

closed 2023-08-17 by sshane