commaai/openpilot #27718: Change fingerprint in lauch_openpilot.sh and get CAN Error: Check Connections

Jackie890621, 2023-03-28:
### Describe the bug

The default fingerprint in ```launch_openpilot.sh``` is *HONDA CIVIC 2016*.
And I wanted to try different cars, so I printed out to see the cars that openpilot supports

[image]

As shown in above image, I choose the first one *HONDA ACCORD 2018*.
And I got the error when running ```bridge.py```

[image]

The accord 2018 and civic 2016 are both belong to HONDA, anything went wrong???

### OS Version

Ubuntu 20.04

adeebshihadeh, 2023-03-28:
This isn't expected to work. Changing cars isn't currently supported in the simulator.

closed 2023-03-28 by adeebshihadeh

Jackie890621, 2023-03-29:
OK~~~Thanks
Can I know which parts of code restrict this?

GHANINEAMAYAS, 2023-06-23:
i have the same problem , can you help me to resolve this problem