commaai/openpilot #27745: libqt_replay.a(consoleui.o): undefined reference to symbol 'nodelay'

hiteshhedwig, 2023-03-30:
### Describe the bug

## STEPS:

Following :
`tools/README.md`

1) 

``` bash
# for Ubuntu 20.04 LTS
tools/ubuntu_setup.sh
```

2) 
``` bash
cd openpilot && poetry shell
```

3)  Now upon running the below command. It produces the error.

``` bash
scons -u -j$(nproc)
```

## ERROR 
```
Generated and written to tools/cabana/dbc/car_fingerprint_to_dbc.json
/usr/bin/ld: tools/replay/libqt_replay.a(consoleui.o): undefined reference to symbol 'nodelay'
/usr/bin/ld: /home/hedwig/anaconda3/lib/./libtinfow.so.6: error adding symbols: DSO missing from command line
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [tools/replay/replay] Error 1
scons: building terminated because of errors.
```

Any idea what possible mistake could be?

### OS Version

Ubunutu 20.04

adeebshihadeh, 2023-03-30:
Thanks for reporting!

closed 2023-03-30 by adeebshihadeh

hiteshhedwig, 2023-04-01:
Hi,
@deanlee thanks alot

Running these two commands below did the job. 
```
sudo apt-get install libncurses5-dev 
sudo apt-get install libncursesw5-dev
```
Thanks~!

adeebshihadeh, 2023-04-01:
@deanlee any idea why this wasn't caught in CI?