commaai/openpilot #27763: Openpilot not showing any view from Carla

keode, 2023-03-31:
### Describe the bug

PC spec: AMD Ryzen 9 5900x + 64 GB Memory + NVIDIA RTX 3070 GPU

Installation: both Carla and openpilot are installed as docker images

- Carla `docker pull carlasim/carla:0.9.13`
- Openpilot `docker pull ghcr.io/commaai/openpilot-sim:latest`

To launch
- Carla (in Terminal 1)
`docker run 
  --privileged 
  --name carla_sim 
  --rm 
  --gpus all 
  -e DISPLAY=${DISPLAY} 
  --net=host 
  -v /tmp/.X11-unix:/tmp/.X11-unix:rw 
  -it 
  carlasim/carla:0.9.13 
  /bin/bash ./CarlaUE4.sh  -nosound -benchmark -fps=20 -quality-level=Low`

- Openpilot (in Terminal 2)

1. `docker run --net=host
  --name openpilot_client 
  --rm 
  --gpus all 
  --device=/dev/dri:/dev/dri 
  --device=/dev/input:/dev/input 
  -v /tmp/.X11-unix:/tmp/.X11-unix 
  --shm-size 1G 
  -e DISPLAY=$DISPLAY 
  -e QT_X11_NO_MITSHM=1 
  -w "/openpilot/tools/sim" 
  -it 
  ghcr.io/commaai/openpilot-sim:latest 
  /bin/bash`
2. `cd /openpilot/tools/sim`
3. `./launch_openpilot.sh`

- Openpilot bridge (in Terminal 3)
1. `docker exec -it openpilot_client bash`
2. `cd /openpilot/tools/sim`
3. `./bridge.py`

Dash cam view is not shown in Openpilot's window. If I press 1 in Terminal 3 where the bridge script is running, it will show "openpilot unavailable, press set to enage"
[image]
[image]

### OS Version

Ubuntu 22.04

### openpilot version or commit

a9d342689cef1a6e21e12a529bcf0ad8cf2dec8e

hugopinho0, 2023-03-31:
I have the same problem and in my case, it only solved the image issue, but I can't activate the autopilot when I press 1. I get the same message from "openpilot unavailable. Press set to engage" 

Also to try to solve the modeld part that appears in red in the logs I have already installed libzmq as they say in [#23666](https://github.com/commaai/openpilot/issues/23666#issuecomment-1045520664) .

Also do you have any other suggestions of what I can do to understand the problem or other things to test?

keode, 2023-03-31:
> I think that passing the ` --high_quality --dual_camera` to `bridge.py` could solve the camera issue as mentioned here #26796

Hi @mkjooriah, thank you. Passing `--high_quality --dual_camera` solves the problem of not having view from Carla, but the problem of "openpilot unavailable. Press set to engage" still persists when I press 1. Any idea?
[image]

keode, 2023-03-31:
> I have the same problem and in my case add the flags `--high_quality --dual_camera`, it only solved the image issue, but I can't activate the autopilot when I press 1. I get the same message from "openpilot unavailable. Press set to engage"
> 
> Also to try to solve the modeld part that appears in red in the logs I have already installed libzmq as they say in [#23666](https://github.com/commaai/openpilot/issues/23666#issuecomment-1045520664) .
> 
> Does anyone have more suggestions of what I can do to understand the problem or other things to test?

I have not installed libzmq though, I'm having the same problem

mkjooriah, 2023-03-31:
> Hi @mkjooriah, thank you. Passing `--high_quality --dual_camera` solves the problem of not having view from Carla, but the problem of "openpilot unavailable. Press set to engage" still persists when [...]

I still have the same issue. I suspected that it could be related to Onnx runtime not yet [officially supporting](https://onnxruntime.ai/docs/execution-providers/CUDA-ExecutionProvider.html) CUDA 12.x https://github.com/microsoft/onnxruntime/issues/13932, so I hardcoded using the `CPUExecutionProvider` instead of `CUDAExecutionProvider` but the issue persisted.

keode, 2023-04-04:
> > > I think that passing the ` --high_quality --dual_camera` to `bridge.py` could solve the camera issue as mentioned here #26796
> > 
> > 
> > Hi @mkjooriah, thank you. Passing `--high_quality --dual_camera` solves the problem of not having view from Carla, but the problem of "openpilot unavailable. Press set to engage" still persists when I press 1. Any idea? [image]
> 
> I still have the same issue. I suspected that it could be related to Onnx runtime not yet [officially supporting](https://onnxruntime.ai/docs/execution-providers/CUDA-ExecutionProvider.html) CUDA 12.x [microsoft/onnxruntime#13932](https://github.com/microsoft/onnxruntime/issues/13932), so I hardcoded using the `CPUExecutionProvider` instead of `CUDAExecutionProvider` but the issue persisted.

I have installed `libzmq` so `modeld` is no longer red in the terminal.
I have checked that I have CUDA version 11.6, but the problem of "openpilot unavailable, press Set to engage" is still not solved.

[image]

hiteshhedwig, 2023-04-05:
In my case it was majorly because the git cloned onnx model was sized 2kb. So, i downloaded again and replaced it in `models/` [folder](https://github.com/commaai/openpilot/blob/master/selfdrive/modeld/models/supercombo.onnx).

adeebshihadeh, 2023-06-25:
Looks like the missing image stream has been resolved. The alert you're seeing is just because you're pressing the wrong button to engage.

closed 2023-06-25 by adeebshihadeh