commaai/openpilot #27791: OpenPilot Unavailable, Waiting for GPS

kwisatzsawyer, 2023-04-02:
### Describe the bug

When I turn on my car, comma 3 says "openpilot unavailable" and "waiting for controls to start" and "Waiting for GPS."

Tried rebooting the comma 3 several times. Tried reinstalling openpilot. Also tried driving around for a few minutes, giving it time. 

Doesn't seem to ever load GPS. In https://useradmin.comma.ai/, every drive has false in the "can" folder. 

### Provide a route where the issue occurs

db2f545ea314b36d|2023-04-02--14-52-35--0

### openpilot version

0.9.1

### Additional info

dongle_id: db2f545ea314b36d

kwisatzsawyer, 2023-04-02:
Double checked the wiring and it appears correct (will describe below). But wanted to note the red panda has a blinking blue LED when my car is off, starts blinking red instead when my car is on.

Order: osb2 -= Ethernet => non-harness black box => USB c => comma main port. Comma 3 aux port => usb-a adapter => red panda => usb-c => car harness => car / camera.

adeebshihadeh, 2023-04-03:
There are no CAN messages on your routes, which indicates the hardware may not be setup properly. Did you follow the setup guide at https://comma.ai/setup/comma-three? Your description sounds correct, but it's ambiguous, so I'd double check with the guide.

adeebshihadeh, 2023-04-03:
Ah, I see you have a support ticked opened and determined this is a hardware issue.

closed 2023-04-03 by adeebshihadeh