commaai/openpilot #27796: calibrationd: add hysteresis on calibration validity

adeebshihadeh, 2023-04-03:
This should prevent mountings that are right on the edge of the bounds from being able to engage sometimes.

sshane, 2023-04-13:
Here's a dongle where this happens occasionally. 9227902ca7eb7bba user states that higher speeds cause the alert more often. (Device is pointed 4 degrees to the right)

adeebshihadeh, 2023-07-14:
This guy's having a bad time with calibration letting him engage sometimes: `5548e173b952a76b|2023-07-08--14-43-34`

beej, 2023-07-15:
> Device is pointed 4 degrees to the right, limit is 3.96

Current understanding is that the Comma wasn't properly seated in the mount at the time, so if comparing with later drives, please keep in mind.

adeebshihadeh, 2024-06-04:
Another user hitting this: `91138069575407d3`.