commaai/openpilot #27839: Ford torque too high on initial activation

ncoop23, 2023-04-07:
### Describe the bug

Bug is that when you are a little off pointing down the lane and start cruise, it sometimes says torque is too high

### Which car does this affect?

Ford Maverick 2023

### Provide a route where the issue occurs

bd37e43731e5964b|2023-04-07--17-46-28--1

### openpilot version

Ford Dashcam 4/?/2023

### Additional info

Not super high priority but it may improve other issues where you go out of lane and it does not warn about torque.

ncoop23, 2023-04-09:
bd37e43731e5964b|2023-04-09--12-02-17--20

Round abouts show the issue and torque coming on too slow also. A few oddities here. Had a warning pop up about hands on the wheel also.

ncoop23, 2023-04-12:
bd37e43731e5964b|2023-04-12--17-29-48--13
Is the best example I can make

ncoop23, 2023-04-16:
bd37e43731e5964b|2023-04-16--12-20-52--8

Another instance. Ramp up after initial activation

ncoop23, 2023-04-27:
@sshane I see you put a commit in for ford steering recently https://github.com/commaai/openpilot/pull/27981  Does this apply?

sshane, 2023-04-27:
No, that is for safety limiting. The last route you posted showed the alert because it wound up taking a sharp turn, then failed to ramp down quick enough, which seems like an appropriate alert for the situation.

sshane, 2023-08-08:
@incognitojam can you reproduce with your Ford? Or does it always steer immediately?

adeebshihadeh, 2024-05-17:
Is this still an issue?

sshane, 2024-05-17:
I think so, Yassine says it takes a few seconds to start steering/ramp up on engage.