commaai/openpilot #27846: 2022 HYUNDAI IONIQ 5 "TAKE CONTROL IMMEDIATELY, Radar Error: Restart the Car"

krvista, 2023-04-08:
### Describe the bug

Comma 3 was functioning normally, 
but as soon as I pressed the button to close the panoramic sunshade, an error message appeared on the screen with a prompt to "TAKE CONTROL IMMEDIATELY, Radar Error: Restart the Car"
**Route ID: 8014970252341f77|2023-04-08--19-43-04.** 

I attempted to reproduce the same error several times.
And I found that it should be cold-started to reproduce this.
I recorded the moment when the same error message appeared on the screen.
**Route ID: 8014970252341f77|2023-04-08--21-21-57.**

There is a possibility that the issue is caused by the devel branch, so I reinstalled Release3 branch.
Then I tried the same thing. 
And the same error occurred, and the CAN BUS was disconnected.
**Route ID: 8014970252341f77|2023-04-09--01-59-52**

Below photos are the warning messages after CAN being diconnected (Orange warning to Red)
[image]
[image]

### Provide a route where the issue occurs

8014970252341f77|2023-04-09--01-59-52

### openpilot version

0.9.1 Release3

### Additional info

2022 HYUNDAI IONIQ 5 4WD (2023MY S.Korean version)
Comma 3 + 0.9.1 Release3 branch (fwd_camera fingerprint of KR version added)
Driven over 1100km with Comma 3

Cx01N, 2023-04-18:
I’ve also experienced this error. Please let me know if there is anything I can provide to help.

krvista, 2023-04-23:
Kia EV6 with a sunroof may also have a similar issue.
When pressing the sunroof open/close button while driving, a message appears on the dashboard asking to check the Front Collision Avoidance System, and openpilot displays the error "CAN Bus Disconnected: Likely Faulty Cable."

sshane, 2023-05-10:
@Cx01N @krvista can you provide a drive where this happens for you? Make sure to upload full logs in connect or leave the device on WiFi so I can request upload them.

krvista, 2023-05-10:
OK. I'll check and upload them soon.

krvista, 2023-05-10:
@sshane I've check and My comma3 already uploaded related drive files except driver facing cameras. 

8014970252341f77|2023-04-09--01-59-52--0

Even though my car and Comma 3 have been parked outside of WiFi range, they are within range of an LTE connection. 
So you can access the videos and logs. 
Please let me know if On WiFi is necessary.

sshane, 2023-05-13:
The sparse logs (not full) are uploaded, but not the full logs. Can you put your device on WiFi to allow them to upload when you get the chance>

adeebshihadeh, 2023-05-15:
It's showing the wrong alert, fixed with e889e09513f5c241dcc53c936966b22e3fe7d9e7. We'll track the real one in https://github.com/commaai/openpilot/issues/27643 

> Kia EV6 with a sunroof may also have a similar issue. When pressing the sunroof open/close button while driving, a message appears on the dashboard asking to check the Front Collision Avoidance System, and openpilot displays the error "CAN Bus Disconnected: Likely Faulty Cable."

I wasn't able to reproduce this on our EV6. If this is reproducible on your car, first try it without the comma three installed. If it's reproducible and **only happens with the comma three installed**, please make a new issue with the full logs uploaded (need to be on WiFi).

closed 2023-05-15 by adeebshihadeh

powers68232, 2023-06-13:
Just received my new Comma 3 and after 3 days of use, this morning I've received these same errors on two different drives. One ended with a red screen and deactivated the cruise. Still waiting for the logs to upload. This is a 19 Chrysler Pacifica.

Abanggo, 2023-07-29:
Same symptom occurs on Lexus NX-300h hybrid 2020.
In addition to openpilot, the same symptoms are occurring in the sub-fork sunnypilot / dragonpilot.

sshane, 2023-08-08:
for reference, that Lexus issue is here: https://github.com/commaai/openpilot/issues/29213