commaai/openpilot #27848: Master-CI branch

kwisatzsawyer, 2023-04-08:
### Describe the bug

Loaded master-ci branch on my Comma 3. It disabled Hyundai's "Forward Collision-Avoidance Assist" (as indicated by that icon lighting up). Happened as soon as I enabled experimental mode. Fix was to wipe the device and reload stock.

### Which car does this affect?

2022 Hyundai Ioniq 5 SEL (w/ HDA II)

### Provide a route where the issue occurs

db2f545ea314b36d|2023-04-08--10-43-32

### openpilot version

0.9.2

### Additional info

[image]

sshane, 2023-04-11:
The problem is that panda went into ELM safety mode, blocking a bunch of messages and causing a fault. I see these errors in the log:

```python
{'errorLogMessage': '{"created": 1680968691.2197044, "ctx": {"daemon": "boardd", "device": "tici", "dirty": false, "dongle_id": "db2f545ea314b36d", '
                    '"version": "0.9.2"}, "filename": "selfdrive/boardd/panda_comms.cc", "funcname": "handle_usb_issue", "levelnum": 40, "lineno": '
                    '140, "msg": "usb error -1 \\"Input/Output Error\\" in bulk_write"}',
 'logMonoTime': 169064965803,
 'valid': True}
{'errorLogMessage': '{"created": 1680968691.2199473, "ctx": {"daemon": "boardd", "device": "tici", "dirty": false, "dongle_id": "db2f545ea314b36d", '
                    '"version": "0.9.2"}, "filename": "selfdrive/boardd/panda_comms.cc", "funcname": "handle_usb_issue", "levelnum": 40, "lineno": '
                    '140, "msg": "usb error -1 \\"Input/Output Error\\" in bulk_write"}',
 'logMonoTime': 169065691949,
 'valid': True}
{'errorLogMessage': '{"created": 1680968691.2244849, "ctx": {"daemon": "boardd", "device": "tici", "dirty": false, "dongle_id": "db2f545ea314b36d", '
                    '"version": "0.9.2"}, "filename": "selfdrive/boardd/panda_comms.cc", "funcname": "handle_usb_issue", "levelnum": 40, "lineno": '
                    '142, "msg": "lost connection"}',
 'logMonoTime': 169073615333,
 'valid': True}
{'errorLogMessage': '{"created": 1680968691.2250469, "ctx": {"daemon": "boardd", "device": "tici", "dirty": false, "dongle_id": "db2f545ea314b36d", '
                    '"version": "0.9.2"}, "filename": "selfdrive/boardd/panda_comms.cc", "funcname": "handle_usb_issue", "levelnum": 40, "lineno": '
                    '142, "msg": "lost connection"}',
```

@adeebshihadeh should we get the device back to look at the panda?

kwisatzsawyer, 2023-04-14:
Let me know if it'd help if I submit any additional info, or if you want me to install & test master-ci branch again. The panda was replaced about a week ago, so it's fresh hardware.

sshane, 2023-04-14:
Is this resolved now?

kwisatzsawyer, 2023-04-16:
Looks like the issue is with the controls on the steering wheel.

Route e/ Experimental Mode + Longitudinal: db2f545ea314b36d|2023-04-16--12-37-43--2

Route in Chill Mode: db2f545ea314b36d|2023-04-16--12-51-17--0

Both with a fresh install of master-ci branch (installed 4/16/2023).

[image]

Notes on image:
1. This button turns on / off ACC (stock car, stock OpenPilot). In Experimental Mode does nothing / sometimes lists errors.
2. Push up to set ACC at current speed, push in to pause/resume (stock, stock). In Experimental Mode: up sets 25 mph, down sets 65 mph.
3. Toggle follow distance. Not tested.
4. LKA assist button. Does nothing in stock branch or master-ci.

adeebshihadeh, 2023-04-19:
This all looks like expected behavior aside from the panda issue @sshane mentioned; if that happens again, email support@comma.ai.

As for the FCA warning light, that's intentionally lit up to highlight that AEB is disabled when OP long alpha is enabled. The cruise buttons also don't exactly match up: OP long doesn't have a follow distance setting and LFA button doesn't do anything.

closed 2023-04-19 by adeebshihadeh