commaai/openpilot #27869: 2023 Toyota Tundra "Car Unrecognized" Fingerprint Capture Included

etc6849, 2023-04-11:
### Describe the bug

I had read that some newer models are using TSK (Toyota Secure Key), but I figured I'd try and see what my 2023 Toyota Tundra non-hybrid model did.  

I'm able to use the Camma 2 in Dashcam mode, and receive the "Car Unrecognized" message.  However, checking the boot logs, I found the fingerprint below.  I would think that this fingerprint wouldn't be there if my ECM was using TSK.  Could it be my new Tundra would work if this fingerprint was added?

`{'key': 'CalibrationParams',
                                      'value': b"\x00\x00\x00\x00\x16\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x009Wj\xb1'\x00\x00\x00\x12\x00\x00\x00"
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x00\x01d\x00\x00\x00\x00\x00\x00\x05\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11\x00\x00\x00d\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1d\x00\x00\x00\x1c\x00\x00\x00!\x00\x00\x00'
                                               b'\x1c\x00\x00\x00\xbb\xda\xb3\xbb\xff\xfe\x7f\xbf\xe4\xa1@:\x00\x00\x00\x00\xad\xe1\x07\xbe'
                                               b'\x00\x00\x80\x9fr\xbc}\xbf\xf6(\x9c?s\xbb}?\x89u\xb5\xbb%\xe1\x07\xbe\x00\x00\x00\x00\xaa9\xd9\xb8'
                                               b'\x95H\x08>\xc6u\xb5\xbb\x00\x00\x00\x00y\x8bG:\xb5\x9d\xc3:\xc8\x0b\xbc;\x00\x00\x00\x00'},
                                     {'key': 'CarBatteryCapacity', 'value': b'7256833'},
                                     {'key': 'CarParams',
                                      'value': b'\x00\x00\x00\x00h\x00\x00\x00\x00\x00\x00\x00\x10\x00\x0e\x00\x82\x00\x00\x00\x00\x00\x80\xbf'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xd4D\xcd\xcc,@\xcd\xcc\xac?\x00\x00PA\x00\x00\x00\x00'
                                               b'\x00@\x1cE\x00$tI\x00$tI\x00\x00\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\xcd\xccL?\x00\x00\x00\x00\x00\x00\x00\x00\x9a\x99\x19>\x00\x00\x00?\x00\x00\x00\xc0\x9a\x99\x19>'
                                               b'\x00\x00\x80?\x00\x00\x00\x00\x00\x00\x00\x00y\x00\x00\x00*\x00\x00\x001\x00\x00\x00*\x00\x00\x00'
                                               b'1\x00\x00\x00\x0c\x00\x00\x001\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b' \x00\x00\x00\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00]\x00\x00\x00\x92\x00\x00\x00'
                                               b'e\x00\x00\x00w\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1d\x01\x00\x00\x0f\x00\x00\x00mock'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80?\x00\x00\x00\x00%\x00\x00\x00'
                                               b'\x0c\x00\x00\x00%\x00\x00\x00\x0c\x00\x00\x00%\x00\x00\x00\x0c\x00\x00\x00%\x00\x00\x00'
                                               b'\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x00\x00'
                                               b'\x0c\x00\x00\x00\x05\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80?\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x80?\x00\x00\x00\x00mock\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x005TFMA5DB4PX071002\x00\x00\x00\x00\x00\x00\x00\x1c\x00\x00\x00'
                                               b'\x01\x00\x01\x00\x02\x00\x00\x00\xd0\x07\x00\x001\x00\x00\x00\x12\x01\x00\x00\x03\x00\x00\x00'
                                               b'\xc4\x07\x00\x00=\x00\x00\x00\x12\x01\x00\x00\x01\x00\x00\x00\xb0\x07\x00\x00I\x00\x00\x00'
                                               b'\x8a\x00\x00\x00\x04\x00\x00\x00\x00\x07\x00\x00M\x00\x00\x00\x8a\x00\x00\x00\x00\x00\x00\x00'
                                               b'\xa1\x07\x00\x00Q\x00\x00\x00\n\x02\x00\x00\x02\x00\x0f\x00P\x07\x00\x00m\x00\x00\x00'
                                               b'\x8a\x00\x00\x00\x03\x00m\x00P\x07\x00\x00q\x00\x00\x00\n\x01\x00\x00\xf1\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x10\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf1\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x84\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x01F15260C35100\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01896630CR730'
                                               b'1\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x048965F3401200\x00\x00\x00\x008A311340500'
                                               b'0\x00\x00\x00\x008965F3402200\x00\x00\x00\x008A3213405000\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x018821F0C13100\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x028646F0C1310'
                                               b'0\x00\x00\x00\x008646G0C15100\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00'
                                               b'\x01\x00\x00\x00\x13\x00\x00\x00\x00\x00\x00\x00'},
                                     {'key': 'CarParamsCache',
                                      'value': b'\x00\x00\x00\x00h\x00\x00\x00\x00\x00\x00\x00\x10\x00\x0e\x00\x82\x00\x00\x00\x00\x00\x80\xbf'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xd4D\xcd\xcc,@\xcd\xcc\xac?\x00\x00PA\x00\x00\x00\x00'
                                               b'\x00@\x1cE\x00$tI\x00$tI\x00\x00\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\xcd\xccL?\x00\x00\x00\x00\x00\x00\x00\x00\x9a\x99\x19>\x00\x00\x00?\x00\x00\x00\xc0\x9a\x99\x19>'
                                               b'\x00\x00\x80?\x00\x00\x00\x00\x00\x00\x00\x00y\x00\x00\x00*\x00\x00\x001\x00\x00\x00*\x00\x00\x00'
                                               b'1\x00\x00\x00\x0c\x00\x00\x001\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b' \x00\x00\x00\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00]\x00\x00\x00\x92\x00\x00\x00'
                                               b'e\x00\x00\x00w\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1d\x01\x00\x00\x0f\x00\x00\x00mock'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80?\x00\x00\x00\x00%\x00\x00\x00'
                                               b'\x0c\x00\x00\x00%\x00\x00\x00\x0c\x00\x00\x00%\x00\x00\x00\x0c\x00\x00\x00%\x00\x00\x00'
                                               b'\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x00\x00'
                                               b'\x0c\x00\x00\x00\x05\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80?\x00\x00\x00\x00\x00\x00\x00\x00'
                                               b'\x00\x00\x00\x00\x00\x00\x80?\x00\x00\x00\x00mock\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
[... cut ...]
                                     {'key': 'GitCommit', 'value': b'5122442b070afd3841e04f374f53da594e0932e3'}, {'key': 'GitDiff', 'value': b''},
                                     {'key': 'GitRemote', 'value': b'https://github.com/sshane/openpilot.git'},
                                     {'key': 'HardwareSerial', 'value': b'9cadb2ee'}, {'key': 'HasAcceptedTerms', 'value': b'2'},`

### Which car does this affect?

2023 Toyota Tundra 

### Provide a route where the issue occurs

bootlogs/73484e3ac5b1b0a7/2023-04-10--23-17-13.bz2

### openpilot version

current

etc6849, 2023-04-11:
Just verified all logs got uploaded for 73484e3ac5b1b0a7|2023-04-10--20-25-02.

[image]

closed 2023-04-11 by sshane

etc6849, 2023-04-22:
Thanks for taking a look.  If anyone has a closed source solution for this, I'm open to beta testing it on my 2023 Tundra.  Feel free to DM etc6849 on Discord.

bitkevin, 2025-12-14:
I own a 2023 Sequoia and got Comma 3X and looking for any solution, how can I contribute to the community?