commaai/openpilot #27909: Replay CAN messages on PC.  boardd_old.py doesnt exist anymore

Eso2000, 2023-04-14:
### Describe the bug

Hello, 
In order to Stream CAN messages to my PC, I would like to use the replay tool. In the ReadME it says:

### Used Instructions:
"""
In order to replay specific route:
MOCK=1 selfdrive/boardd/tests/boardd_old.py
In another terminal:
tools/replay/replay <route-name>
"""
but the boardd_old.py doesnt exist anymore. Is there a replacement for this?
Thank you in advance!

### OS Version

Ubuntu 20.04

MoreTore, 2023-04-17:
> ### Describe the bug
> 
> Hello, In order to Stream CAN messages to my PC, I would like to use the replay tool. In the ReadME it says:
> ### Used Instructions:
> 
> """ In order to replay specific route: MOCK=1 selfdrive/boardd/tests/boardd_old.py In another terminal: tools/replay/replay """ but the boardd_old.py doesnt exist anymore. Is there a replacement for this? Thank you in advance!
> ### OS Version
> 
> Ubuntu 20.04
> ### openpilot version or commit
> 
> _No response_
> ### Additional info
> 
> _No response_

Just run Cabana in streaming mode

adeebshihadeh, 2023-04-19:
Fixed this up in #27975; feel free to open a PR to further improve it.

closed 2023-04-19 by adeebshihadeh