commaai/openpilot #27916: UI watchdog reset when scrolling device after incompatible NVME installation/removal.

MoreTore, 2023-04-15:
### Describe the bug

Today I tried to install an nvme drive into my C3 but it caused many issues and later figured out that the nvme must be a specific type. Now after removing the ssd and resetting the device (it corrupted stuff) I am getting UI watchdog resets whenever I try to scroll on any menu. I tried to factory reset again, but now I am stuck on the terms and conditions agreement unable to scroll without the UI crashing. 

### Provide a route where the issue occurs

164080f7933651c4

### openpilot version

NA

Verylukyguy, 2023-04-15:
https://github.com/commaai/agnos

MoreTore, 2023-04-15:
Yea I'm going to reflash it

closed 2023-04-15 by MoreTore