commaai/openpilot #27986: GM Camera ACC: send cancel to camera to turn off stock cruise when OP long

sshane, 2023-04-20:
Camera sends alerts over GMLAN which we can't control, so you'll get certain alerts like "hill too steep" and other strange alerts that are caused by mismatches between the camera thinking it's engaged (it doesn't check the ECM status :facepalm:) when the ECM isn't (like if OP catches an fault from the brake pressure issue at a stop).

Send a cancel button to the camera if it is enabled with OP long to silence some of these alerts

- [ ] Check everything still works as expected

jud336, 2023-04-24:
Do not work

jud336, 2023-04-24:
Yes I work[](url)

jud336, 2023-04-24:
[](url)

sshane, 2023-04-24:
@jud336 do you have any comment for the issue?

TurboCE, 2023-08-15:
Is this issue the same as what I experienced? 
When using E2E OP long, it fails to operate.

5085c761395d1fe6|2023-08-15--22-16-33
(ex. Segment 7, 8, 9)