commaai/openpilot #28023: Dash Cam unrecognized

DanSantaCruz, 2023-04-25:
### Describe the bug

2023 Hyundai Santa Cruz, Dash Cam not recognized

### Which car does this affect?

2023 Hyundai Santa Cruz

### Provide a route where the issue occurs

all

### openpilot version

Dont know

sshane, 2023-04-26:
Can you provide your dongle id? It will be in device settings

DanSantaCruz, 2023-04-26:
Good morning,

Thank you for your response. Dongle and serial in attachment.

It is 617bb607e012533e

Serial Number bad7504c

Thank you,
Dan

sshane, 2023-04-26:
Hi, your car's fingerprints were added here 2 months ago, but they are not in release yet: https://github.com/commaai/openpilot/pull/27395

Simply install the nightly branch to get up and running!

Installer URL: https://installer.comma.ai/commaai/nightly

closed 2023-04-26 by sshane