commaai/openpilot #28046: OP-Long HKG ICE CAN-FD

ebg1223, 2023-04-27:
@sunnyhaibin had OP-Long on ICE CAN-FD working almost a year ago, and another branch had support almost 8 months ago. There was an impending safety refactor the comma team was planning to work on directly, and understandably, the team did not want to merge a patch prior to this in order to support ICE vehicles. However, this refactor seems to have been dropped pretty far down the priority list, if it is still planned.

A significant number of users are running with a sub-optimal experience, not due to any technical limitations. Additionally, I personally know 2 potential buyers who want to purchase(one Santa Cruz, one Tucson), but want to wait until it is "fully supported" before spending the money, especially with the amazing demos the comma team has put out on what is possible with full control. I am sure there are some others like this out there as well.

Hoping the comma team will permit a patch to support ICE CAN-FD on OP-Long, and make the large panda safety refactor a separate, long-term goal. I believe the additional complexity was not significant. Hoping to not let perfect be the enemy of a good solution for a fairly decent size user base. Thank you in advance for your consideration!

BrandonHacks, 2023-04-27:
I would like to echo what you’ve said. I appreciate the comma team and am thrilled with the experience this far. I actually purchased a new vehicle as an upgrade for comma and have been disappointed that I’m not able to utilize all of the features that my Sportage is capable of with OP.

ebg1223, 2023-04-27:
@BrandonHacks Agree completely! Not trying to detract from all the hard work at comma and the amazing product!

gaanthony, 2023-04-30:
I've been waiting for this ever since I had a taste of it on the original PR for my ICE Kia Sportage 2023 but lost it in the merge.  I've waited patiently for months wondering why a software change is taking so long.

gomtings, 2023-05-04:
May I know what ICE CAN-FD is? Is it a new project?

BrandonHacks, 2023-05-04:
> May I know what ICE CAN-FD is? Is it a new project?

ICE is internal combustion engine(so gas models). CAN-FD is the communication protocol that is used between the car and comma to send commands. If you have a red panda, then you have CAN-FD.

gomtings, 2023-05-16:
Think you

BrandonHacks, 2023-07-10:
Sure would be nice to hear something from comma team.. anything?

adeebshihadeh, 2023-08-07:
Closing since this isn't a bug. We're aware of this, and it's on the TODO list. Feel free to open a GitHub Discussion to discuss this further.

closed 2023-08-07 by adeebshihadeh