commaai/openpilot #28047: Toyota ICE 8-Speed and CVT winds up/jerks clutch when sending negative ACC values

rockerrock137, 2023-04-27:
### Describe the bug

Current ACC for Toyota often sends negative values in the ACC command which results in:
* Torque converter unlocking, and in some cases, downshifts on 8 speed automatic transmissions
* Clutch engagement and ratio changes for CVT

This issue is very apparent when running in Experimental mode as the ACC signal is often +/- zero point.

Obviously, constantly changing the CVT ratio and engaging/disengaging torque converter lockup isn't good for the transmission long term. It's also uncomfortable for passengers as there is a constant "catch" feeling from the drivetrain. More often in experimental, less often in chill, but still noticeable.

I've experimented with increasing the deadzone, which seems to work well, but ideally, we'd have a deadzone on the zero crossing boundary for negative values. Consideration should be made as to if this applies to only ICE vehicles vs hybrid, as hybrid regeneration behavior may be smoother and/or not noticeable.

An A/B test could be as simple as comparing stock ACC behavior route to an OP controlled route. It would also be beneficial to also log engine RPM, or some other value representative of transmission operation.

### Which car does this affect?

Toyota ICE Automatic Transmissions

### Provide a route where the issue occurs

https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/interface.py#L233

### openpilot version

5a4452e3c636d8e4d8ea9b4456f5821c943342b2

asdfasd2312, 2023-04-27:
I concur. 

https://discord.com/channels/771493367246094347/771495215570747403/1100872546863296602

ntegan1, 2023-05-19:
I have a TSSP ICE CVT corolla and would definitely like OP to interact with the powertrain more closely to how stock ACC does as its longitudinal policy is designed/tuned for each car.

One major difference I've noticed is that on downhills, my stock ACC will coast with engine braking for at least a few dozen seconds (unless there's a lead car or the set speed is decreased) before applying friction braking.