commaai/openpilot #28050: controlsd: can get stuck if ignition is quickly cycled

sshane, 2023-04-27:
I assume it's something to do with the blocking OBD multiplexing setting

adeebshihadeh, 2023-04-27:
Are you sure it's not the safety setter thread in boardd?

sshane, 2023-04-27:
Yeah, I think the two got out of sync by the ignition cycling.

closed 2023-05-10 by sshane

sshane, 2023-05-10:
simple explanation of the problem: thermald takes a few seconds before it sets `started=False` after ignition falls, so you can exploit that to achieve this bug:

1. while openpilot is starting up, turn ignition off and on quickly so safety setter thread gets reset.
2. controlsd will then write a param like `FirmwareQueryDone` to make the boardd thread move onto the next step
3. thermald finishes setting power save mode and sets started=False
4. controlsd dies
5. thermald finishes setting power save mode and sets started=True
6. controlsd starts and tries to interact with boardd but it's in the next step or the relay is open and thread is dead depending on how far controlsd got before it died