commaai/openpilot #28058: Unexpected disengage - 2020 Hyundai Santa Fe

tylermilner, 2023-04-28:
### Describe the bug

While driving on some city streets, I had openpilot unexpectedly disengage on me. I believe an error message popped up on the dash to the effect of "Smart Cruise Control conditions not met", which normally only pops up if you try to engage when stock Hyundai SSC isn't available (e.g. foot on brake, etc.). However, I didn't hit the brake or even touch the steering wheel as I had only just engaged openpilot a few seconds prior.

I was able to reengage openpilot a few seconds later down the road and it worked just fine so I'm not sure why it stopped. This is the only time I've encountered this issue on this car since June of 2022. Happens at about the `18:08:41` mark in `e56ffe1d4cbc4f6c|2023-04-15--17-32-21--36` on openpilot `0.9.1` (`devel` branch).

### Which car does this affect?

Hyundai Santa Fe 2020

### Provide a route where the issue occurs

e56ffe1d4cbc4f6c|2023-04-15--17-32-21--36

### openpilot version

0.9.1 (devel)

### Additional info

Here is a clip of the event. You can see I engage openpilot about 4 seconds in and then the unexpected disengagement happens at about the 15 second mark:
https://user-images.githubusercontent.com/1720304/235063326-17a699c7-1097-44dd-b49b-7d3f89fd27fc.mov

sshane, 2023-05-10:
It looks like you may have hit some uneven ground and ABS activated as it was braking, also indicated to be true by the `ABS_ACT` signal that went high around this time: 
[image]

sshane, 2023-05-10:
Thanks for the issue, but closing since this is a duplicate of https://github.com/commaai/openpilot/issues/23601

closed 2023-05-10 by sshane