commaai/openpilot #28061: New Honda Harness

vanillagorillaa, 2023-04-28:
- [x]  which car is this for? 2023 Honda Pilot
- [ ]  schematic
- [ ]  beta sent to tester
- [ ]  in shop

Reported on Discord by user "Kevmaninc". Unknown connector part number. Will have to grab wiring diagram from Honda Techinfo. Unknown camera module. No marking on underside of camera. Will assign proper harness name once more information is obtained.

Kevmaninc, 2023-04-29:
[image]
[image]

This is what the connector looks like.

vanillagorillaa, 2023-04-30:
@Kevmaninc Can you please provide more clear photos of both front and back of the connector? Usually you can see markings in the plastic that can help determine the part number.

vanillagorillaa, 2023-04-30:
This TE connectivity connector looks like it may be a close match, https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/1318386-2/9925408

Kevmaninc, 2023-04-30:
> This TE connectivity connector looks like it may be a close match, https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/1318386-2/9925408

It’s close but when you take a closer look, the distance between the top and bottom pin holes is a bit further apart.

Kevmaninc, 2023-04-30:
I think I found the number you need for the part. See the pics
[image]

Kevmaninc, 2023-04-30:
[image]
[image]
[image]
[image]

Kevmaninc, 2023-04-30:
Here is a link to Chinese site I found for that connector.

https://www.yixiandianzi.com/lianjieqi/309.html

vanillagorillaa, 2023-04-30:
> GT25-16DS-HU

Yes, this appears to be the correct part number. This connector can be purchased on Mouser, no need for a Chinese clone.

Kevmaninc, 2023-04-30:
OK so what you are saying is that I just need a male and female version of this and I should wire them up myself? I noticed that the old pilot connector has 11 wires while the new one has only 6. I guess I just have to wire them up to the correct slots and it should work?

vanillagorillaa, 2023-06-14:
[image]
Sharing @Kevmaninc 's picture of the camera from discord for documentation

Mr-Peter-Ng, 2023-07-15:
@Kevmaninc i also own a 2023 Pilot Touring, I was wondering if you’ve found something (adapter) to help me connect my comma 3 to my vehicle. I currently also have the Nidec harness that isn’t a proper fit.

Kevmaninc, 2023-07-15:
So yeah we actually got the connector made and working, however the software is having issues so we are still working on figuring this out.Sent from my iPhone

Mr-Peter-Ng, 2023-07-15:
Would you be kind enough to share a link on where I could get the correct connector & possibly some photos of your current wiring so I’d be able to rig it up myself. Unless you have an extra I could purchase that’s already pre wired. Sent from my iPhone

Kevmaninc, 2023-07-15:
The wire is not available for purchase yet because there is no official support. If you want to join the testing team to see if we can get this thing running maybe vanillagorilla from the comma team can send you the wire and see if between my pilot and your pilot we can get things running. Do you have comma 3?Sent from my iPhone

Mr-Peter-Ng, 2023-07-15:
Yes I’ll reach out to vanillagorilla Also yes I have the comma 3Sent from my iPhone

Kevmaninc, 2023-07-15:
Hopefully between our two Hondas and the comma team we can get it running.Sent from my iPhone

Mr-Peter-Ng, 2023-07-15:
@vanillagorillaa hello, my name is Peter I’m interested with getting into contact to possibly joining the comma beta testing group to help provide feedback for the comma 3 on my 2023 Honda Pilot Elite. But first I’m looking at acquiring the proper harness/adapter.

Kevmaninc, 2023-07-16:
Peter please join the DISCORD channels and friend me on Discord: @Kevmaninc

OmarWazzan, 2023-09-07:
hey hey - any update on this? thinking of buying a ‘23 accord for Openpilot

Kevmaninc, 2023-09-07:
As of now COMMA is not prioritizing this and it is their lowest priority.
At this point it is unknown as to when or if they can get this up and
running.

sshane, 2023-09-07:
Is anyone working on the port for these vehicles? Are there any routes with rlogs I can take a look at?

vanillagorillaa, 2023-09-07:
> Is anyone working on the port for these vehicles? Are there any routes with rlogs I can take a look at?

Sent you routes via DM a while ago

sshane, 2023-09-08:
See discussion about the port here: https://github.com/commaai/openpilot/discussions/29825

cppctek, 2023-09-14:
are you needing any one else to test it on a 2023 honda accord? I have one and I would love to test it at my own risk and help out with the process. I have a background in both automotive and tech. I would love to help if you would allow me the opportunity. I am very hardware savy.

nate-z, 2023-09-21:
Let me know if anyone needs testing done or support for the 2023 CR-V Hybrid Touring, I am more than happy to help in any way I can to get this car ported and supported.

Mr-Peter-Ng, 2023-09-21:
I’m not quite sure testing is the issue, I think we just need to find more people to help solve the software and figure out what’s wrong with the code & why it’s continually being rejected by the vehicle, & all the dashboard error lights. Until this is solved, I don’t think having more vehicles will make a difference. I will say the new refreshed 2022 civic has the same interior steering wheel & was updated to the new body style by Honda & currently that is the only known vehicle to work with the newest Hondas. Of the new refreshed/redesigned generations of pilots, accords & crv’s, Civic is the only known vehicle to work with openpilot.

smtermini, 2023-09-23:
I, too, just got a new car. I was driving a 2016 Civic and loved this add-on, and I just got an Accord 2024; if you need any testers, please feel free to let me know. I'm happy to test.

Jgw868, 2023-10-07:
I too have a 2023 honda accord hybrid touring, and a comma 3x, would really love to help crack the code get this one running if someone can catch me up to speed

vanillagorillaa, 2023-10-24:
Logs from a test user show that Honda is now using CAN-FD on the radar bus. This will mean that either a Red Panda is needed (if using a comma 3), or a 3X will be needed. 

This resolves the unknown cause of the vehicle dashboard alerts. I will move this new harness into production and will have on the shop for purchase in the coming weeks.

cwmayes, 2023-11-14:
@vanillagorillaa 
Is the plan to still add the harness to production or is this still being worked?  Thanks

akshat-ja, 2024-01-15:
I'm getting a CR-V 2024 and will test out the 3X on that soon.

Bensoncheong626, 2024-02-24:
i'm getting a stepwgn 2022 use new Bosch camera , how can i get the cable?

shahrk, 2024-04-19:
> I'm getting a CR-V 2024 and will test out the 3X on that soon.

@akshat-ja Were you able to get it to work?

tomngu96, 2024-05-05:
For everyone following along at home, I don't forsee the bosch-c being added to the shop in the near future but we have made some fun discoveries and several people have successfully built their own harness and got the 11th gen accord running with the bosch-c harness. It seems the pilot is also functional.

I documented some things like how to build the harness here https://github.com/commaai/openpilot/wiki/Honda-Bosch%E2%80%90C

I am selling some of the connectors I built on the comma discord @Bensoncheong626 and whoever else is interested though I am not sure if your car is actually using the newest bosch setup.

Jgw868, 2024-05-05:
hell yeah brother!! I will have to build my own harnesss then if someone
can point me in the right direction with any parts I may need or any
instructions if available, I would love to get my 2023 honda accord hybrid
touring up and running!! my comma x has been sitting mounted in the car
since i got it in august last year...havent fully completed the install due
to no harness. If anyone can give me a few pointers I will be eternally
grateful!!!

thank you!!

Jonathan Willemain
***@***.***

shahrk, 2024-05-05:
@tomngu96 How much are you selling for? I am not on the discord but I am interested for making comma work on my 2023 CR-V

tomngu96, 2024-05-05:
@Jgw868 Here's the guide I created with pictures, videos, and a parts/tools list so that anyone can make their own if they want https://github.com/commaai/openpilot/wiki/Honda-Bosch%E2%80%90C

@shahrk I've emailed you so as to not have an unrelated and long discussion on this issue. The synopsis is $100 to match comma's price. I'll give steep discounts to people who've donated to the cart port before today or who have contributed to comma/op

Jgw868, 2024-05-05:
@tomngu96 Awesome, thanks for the info! I think I donated $100 to the
gofundme for the port and i think it's worth every penny

EfrB, 2024-07-08:
> For everyone following along at home, I don't forsee the bosch-c being added to the shop in the near future but we have made some fun discoveries and several people have successfully built their own [...]

Is there any still available?
I'm looking for hardness for my Honda Odyssey 2024

I'm not on discord.

tomngu96, 2024-07-10:
I believe the odyssey is not bosch c yet but could be wrong. I believe at
least the 23 odyssey is a bosch A from chatting with someone on discord

Here's an odyssey thread
https://github.com/commaai/openpilot/discussions/2349?sort=new#discussioncomment-8248688

You can send pictures of your harness to me on discord or via email and I
can try to help steer you in the right direction. I prefer discord as thats
also where you'll likely find people with odysseys

My email is my github name @gmail.com

EfrB, 2024-07-10:
> I believe the odyssey is not bosch c yet but could be wrong. I believe at
> least the 23 odyssey is a bosch A from chatting with someone on discord
> 
> Here's an odyssey thread
> https://github.com/commaai/openpilot/discussions/2349?sort=new#discussioncomment-8248688
> 
> You can send pictures of your harness to me on discord or via email and I
> can try to help steer you in the right direction. I prefer discord as thats
> also where you'll likely find people with odysseys
> 
> My email is my github name @gmail.com
> 
> 
> 
> > For everyone following along at home, I don't forsee the bosch-c being
> > added to the shop in the near future but we have made some fun discoveries
> > and several people have successfully built their own harness and got the
> > 11th gen accord running with the bosch-c harness. It seems the pilot is
> > also functional.
> >
> > I documented some things like how to build the harness here
> > https://github.com/commaai/openpilot/wiki/Honda-Bosch%E2%80%90C
> >
> > I am selling some of the connectors I built on the comma discord
> > @Bensoncheong626 <https://github.com/Bensoncheong626> and whoever else is
> > interested though I am not sure if your car is actually using the newest
> > bosch setup.
> >
> > Is there any still available?
> > I'm looking for hardness for my Honda Odyssey 2024
> >
> > I'm not on discord.

You're right. It will need the Bosch A with some software adjustments.