commaai/openpilot #28172: camerad: persistent crash after restarting openpilot many times

sshane, 2023-05-12:
hit this while restarting Bronco many times in a row to test a signal. camerad crashed every subsequent startup 54827bf84c38b14f|2023-05-11--22-06-44

Related errorLog:

```
{'errorLogMessage': '{"msg": {"tombstone": "system/camerad/camerad - Signal: 6 (SIGABRT) -  __GI_raise  at ../sysdeps/unix/sysv/linux/raise.c:50"}, '
                    '"ctx": {"daemon": "tombstoned", "dongle_id": "54827bf84c38b14f", "version": "0.9.2", "origin": "github.com/commaai/openpilot", '
                    '"branch": "master", "commit": "432f316e8df9fd614f784f0b5b3b445d14cc5f8e", "dirty": false, "device": "tici"}, "level": "ERROR", '
                    '"levelnum": 40, "name": "swaglog", "filename": "tombstoned.py", "lineno": 125, "pathname": '
                    '"/data/openpilot/selfdrive/tombstoned.py", "module": "tombstoned", "funcName": "report_tombstone_apport", "host": "tici", '
                    '"process": 121853, "thread": 547503089488, "threadName": "MainThread", "created": 1683868011.5698807}',
```

sshane, 2023-07-22:
hit again, more verbose:

```
comma@tici:/data/openpilot/system$ ./camerad/camerad
system/camerad/cameras/camera_util.cc: VIDIOC_CAM_CONTROL error: op_code 266 - errno 19
dt is 2c
system/camerad/cameras/camera_util.cc: VIDIOC_CAM_CONTROL error: op_code 266 - errno 19
dt is 2c
system/camerad/cameras/camera_util.cc: VIDIOC_CAM_CONTROL error: op_code 266 - errno 19
dt is 2c
system/camerad/cameras/camera_util.cc: VIDIOC_CAM_CONTROL error: op_code 258 - errno 22
camerad: system/camerad/cameras/camera_qcom2.cc:726: void CameraState::camera_open(MultiCameraState *, int, bool): Assertion `isp_dev_handle_' failed.
Aborted (core dumped)
comma@tici:/data/openpilot/system
```

adeebshihadeh, 2023-12-17:
@deanlee want to look into this? I suspect it's not cleaning up some state on one of the exits.

deanlee, 2023-12-18:
I have not been able to reproduce this issue so far. Does it still happen after #30642 ?

Also I think it's better to zero memory before returning from MemoryManager::alloc https://github.com/commaai/openpilot/pull/25728 , so we can avoid passing invalid parameters(EINVAL) to ioctl.

adeebshihadeh, 2023-12-18:
Yes, we still see it in CI sometimes. I suspect it’s caused by interrupted shutdowns. 

We can reopen and merge that one.

adeebshihadeh, 2024-10-12:
Got a theory on this after working on the debayer stuff:

The 845 has 2x IFEs and 4x IFE lites. I suspect previous camerad runs are leaving the IFEs in a bad state, and it's hard to repro because we have three extra to fall back to.

closed 2024-12-16 by adeebshihadeh