commaai/openpilot #28183: Toyota Highlander Hybrid 2023 unrecognized

patdeg, 2023-05-13:
### Describe the bug

Hi team,
I just bought a Toyota Highlander Hybrid Limited AWD 2023 and my comma 3 (version 0.9.1-release / dashcam 3 / 6b6f043 / Feb 23) does not recognize the car (message "Car Unrecognized"). Dongle ID 73289428cd1e4a51. Serial fa91f581.
Do you know how I could make my car work?
Thanks
Patrick

### Which car does this affect?

Toyota Highlander Hybrid Limited AWD 2023

### Provide a route where the issue occurs

73289428cd1e4a51

### openpilot version

0.9.1

sshane, 2023-05-15:
You're driving on dashcam software, I recommend installing openpilot to get started. Factory reset and enter this URL during setup:

https://openpilot.comma.ai

patdeg, 2023-05-15:
Thanks, I uninstall the software, reset the device and install a custom
build using the URL you provided.

I did get a new training module on what it can and can not do.

However I have the same issue with "Car Unrecognized".

Any suggestions?

Thanks

Patrick

sshane, 2023-05-15:
Oh your car's fingerprints must have been added recently to master, so they are not yet in release. Try this installer URL, but note that we have a new release we're pushing soon, so you can switch back to `release3` then. For now, use this:

https://installer.comma.ai/commaai/master-ci

sshane, 2023-05-16:
Closing since your car is supported on master, let me know if you have any trouble installing master-ci!

closed 2023-05-16 by sshane