commaai/openpilot #28197: New cabana doesn't display acceleration correctly

jackking444, 2023-05-15:
### Describe the bug

New cabana doesn't display acceleration correctly 
 SG_ ACS_Sollbeschl : 24|11@1+ (0.005,-7.22) [-7.22|3.005] "Unit_MeterPerSeconSquar" XXX

[image]

[image]

### Provide a route where the issue occurs

e12eb71c201af95a|2023-03-24--13-56-39

### openpilot version

0.9.2

sshane, 2023-05-16:
works for me, @deanlee want me to close?

[image]

deanlee, 2023-05-16:
This is not normal. even older versions should not have this issue. I have not encountered this problem yet during my testing of this route. Could you provide more information about this bug?

jackking444, 2023-05-16:
No, used from opendbc repo and I restore it(when I remove some signals in cabana)
Now git status in data/openpilot/opendbc
git status
HEAD detached at 9a1de83
nothing to commit, working tree clean

[vw_golf_mk4.txt](https://github.com/commaai/openpilot/files/11484922/vw_golf_mk4.txt)

jackking444, 2023-05-16:
> are you still experiencing this issue now, after restoring opendbc/vw_golf_mk4 to its original state? (You may also need to make sure there are no opendbc/vw_glof_mk4.dbc.tmp file automatically saved after the last modification.)
Yes, I am. 
I install OP and tools by guide:
https://github.com/commaai/openpilot/tree/master/tools

but I have Ubuntu 22.04.2

jackking444, 2023-05-16:
I will try and come back

jackking444, 2023-05-17:
Doesn't work, and I decided to rebuild OP and cabana doesn't build
In file included from tools/cabana/dbc/dbcfile.cc:1:
./tools/cabana/dbc/dbcfile.h:11:24: error: base class has incomplete type
class DBCFile : public QObject {
                ~~~~~~~^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:52:7: note: forward declaration of 'QObject'
class QObject;
      ^
tools/cabana/dbc/dbcfile.cc:11:67: error: type 'QObject' is not a direct or virtual base of 'DBCFile'
DBCFile::DBCFile(const QString &dbc_file_name, QObject *parent) : QObject(parent) {
                                                                  ^~~~~~~
tools/cabana/dbc/dbcfile.cc:26:82: error: type 'QObject' is not a direct or virtual base of 'DBCFile'
DBCFile::DBCFile(const QString &name, const QString &content, QObject *parent) : QObject(parent), name_(name), filename("") {
                                                                                 ^~~~~~~
3 errors generated.
scons: *** [tools/cabana/dbc/dbcfile.o] Error 1

jackking444, 2023-05-17:
I did git pull and build it. But original problem still has.
Can you test with my route e12eb71c201af95a|2023-03-24--13-56-39?

deanlee, 2023-05-18:
`cabana "e12eb71c201af95a|2023-03-24--13-56-39" `
[image]
[image]

It looks good to me. it would be more helpful if you provide some more detailed information such as record a video from starting Cabana to when the issue occurs.

jackking444, 2023-05-18:
I can added a screenshot with issue from new build of cabana and make a video from starting Cabana.
I don't know what I give more. 
My guess maybe cabana doesn't zoom in(rounding) value of this signal.
It can be connected with libcapnp-0.7.0.so? 
Because plotjuggler want this lib, but ubuntu 22.04 has libcapnp-0.8.0.so.
Also I try to run cabana with debug mode.

jackking444, 2023-05-18:
May be, I don't know, but when cabana start it created dbc.tmp and in this file Factor is zero, but in original file has a value.
If I change factor in cabana interface it change at dbc.tmp file and cabana is starting show correct.
Example on an screenhot:
ACC_System ACS_Sollbeschl I changed it in the interface
ACS_zul_Regelabw I don't do anything
[image]

deanlee, 2023-05-18:
>  but when cabana start it created dbc.tmp and in this file Factor is zero

No, Cabana does not generate .tmp files at startup. This file is only generated after you have edited dbc file in Cabana. I think it's likely that this issue occurred because you accidentally edited the DBC file. and then the automatically saved file was not deleted for some reason (perhaps because Cabana did not exit properly and clear all tmp files).

deanlee, 2023-05-18:
> An Idea in /home/jack/openpilot/tools/cabana/dbc/dbcfile.cc may be checks doesn't passed? .arg(sig->factor, 0, 'g', std::numeric_limits::digits10) .arg(sig->offset, 0, 'g', std::numeric_limits::digits10)

I believe there are no issues with this part of the code, and this issue seems to be caused by accidental editing. If it works fine after deleting 'vw_glof_mk4.dbc.tmp', and subsequent editing, saving, and loading of dbc are also without any problems, then there should be no other code-related issues.

jackking444, 2023-05-18:
> No, Cabana does not generate .tmp files at startup. This file is only generated after you have edited dbc file in Cabana. I think it's likely that this issue occurred because you accidentally edited [...]

You right, but cabana doesn't load factor from DBC in my case:

https://github.com/commaai/openpilot/assets/11144568/6e33478d-82c0-4d69-ab66-818b2627b645

jyoung8607, 2023-05-18:
As a test, use a text editor to make a direct edit to opendbc/vw_golf_mk4.dbc and see if it shows up. For example, change the offset from -7 to -6.

deanlee, 2023-05-18:
Can you upload the dbc file you loaded so I can test it?

In theory, there shouldn't be any issues with parsing the dbc file because the parsing is done in the opendbc module. And this part of the code is very stable.

jackking444, 2023-05-18:
> Can you upload the dbc file you loaded so I can test it?
> 
> In theory, there shouldn't be any issues with parsing the dbc file because the parsing is done in the opendbc module. And this part of the code is very stable.

Maybe Qt loading? I'm AFK for while, but I will upload

jackking444, 2023-05-18:
[vw_golf_mk4.dbc.txt](https://github.com/commaai/openpilot/files/11511315/vw_golf_mk4.dbc.txt)
[vw_golf_mk4.dbc_old.txt](https://github.com/commaai/openpilot/files/11511316/vw_golf_mk4.dbc_old.txt)
dbc_old.txt it's original from repo 
dbc.txt it's from discord to test

jackking444, 2023-05-18:
I figured out it's "Locale" problem.
signalview.cc:  double_validator->setLocale(QLocale::C);  // Match locale of QString::toDouble() instead of system
I changed my system Locale to en_US.UTF-8... I just live in wrong country:)

jackking444, 2023-05-21:
It's 100 percent reproduced if I change system locale.
I think it's a some kind of bug or specific behavior.

deanlee, 2023-05-21:
which locale will cause this issue?

jackking444, 2023-05-23:
My locale:
`locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=ru_RU.UTF-8
LC_TIME=ru_RU.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=ru_RU.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=ru_RU.UTF-8
LC_NAME=ru_RU.UTF-8
LC_ADDRESS=ru_RU.UTF-8
LC_TELEPHONE=ru_RU.UTF-8
LC_MEASUREMENT=ru_RU.UTF-8
LC_IDENTIFICATION=ru_RU.UTF-8
LC_ALL=
`
I make a video with with demo route with my locale with problem and use US locale everthing is ok:
https://drive.google.com/file/d/1K75e1u6i-fb51VUeMvxkAFB8qwDss1ey/view?usp=sharing

closed 2023-05-24 by adeebshihadeh