commaai/openpilot #28215: Stuck on "Registering Device"

swieder711, 2023-05-17:
### Describe the bug

After updating to latest DP release today on my C3, my device got stuck on "Registering Device" screen. Could not reset to factory default on boot screen. Many others reported similar problem on Discord.

### Provide a route where the issue occurs

serial: 2246c0fa

### openpilot version

Latest 

### Additional info

Thanks

swieder711, 2023-05-17:
Wrong thread for a Dragon Pilot problem.

closed 2023-05-17 by swieder711

EthanChou860314, 2024-03-08:
When I power on, my device got stuck on "Registering Device" screen. 
I dis connect the power, and then connect power. It's still stuck on "Registering Device" screen. 
How can I reset my device?

What is Dragon Pilot problem?