commaai/openpilot #28324: GPS signal error - always displayed

sunilm1992, 2023-05-28:
### Describe the bug

My car has a GPS signal issue with the comma3, having returned the device already and being given another one I think the most likely cause is the heating elements in my windscreen
When this GPS error comes up it says 'hardware malfunction if sky visible' but this blocks the entire screen mostly and blocks useful information

please can we change this warning to a smaller one so that it doesn't block the screen / only show it for 30 seconds or something?

### Provide a route where the issue occurs

see above

### openpilot version

0.9.2

haraschax, 2023-05-29:
Can you post a route? And a picture of your setup?

haraschax, 2023-06-06:
Not actionable without a route

closed 2023-06-06 by haraschax