commaai/openpilot #28361: `resumeRequired` does not display in UI if `belowSteerSpeed`

nworb-cire, 2023-06-01:
On cars without steer to zero, the `belowSteerSpeed` message takes priority over `resumeRequired` when at a standstill. Displaying the latter message at a standstill makes it clear to the driver that their input is required, and its absence may lead to confusion about why the vehicle is not moving.

I have confirmed with several people that this is the case for GM, and am checking on Toyota (those are the only two makes that appear to set the `resumeRequired` message).

cydia2020, 2023-06-02:
Just pasting my messages from discord as I've seen this issue as well

> I can confirm I've noticed it as well. The "Press resume to exit standstill" alert doesn't seem to work on my Prius which can go down to 0 but still requires a accelerator pedal tap or a resume button press to move again.

> IMHO I don't think it's related to any of those mods (SDSU, pedal, SnG hack), except the Pedal, as
```python
if self.CP.openpilotLongitudinalControl:
      if ret.cruiseState.standstill and not ret.brakePressed and not self.CP.enableGasInterceptor:
        events.add(EventName.resumeRequired)
```
> I don't remember seeing this alert at all, even back when I didn't have a pedal.
> That being said, to add more information - I do have both an SDSU and a pedal attached.

sshane, 2023-06-02:
It was added to Toyota about 7 months ago, did you have the pedal since then? That could be why you haven't seen it. https://github.com/commaai/openpilot/pull/26585

sshane, 2023-06-02:
Trying a higher priority + only shows when openpilot actually wants to go: https://github.com/commaai/openpilot/tree/gm-show-resume-alert

nworb-cire, 2023-06-02:
Thanks for looking at this so quickly! 

I believe that the `CC.cruiseControl.resume` signal is a bit noisy, especially in experimental mode. This change appears to work correctly when a lead is detected, but at a stoplight with no lead, it was rapidly alternating between the two alerts, making the `AudibleAlert.prompt` sound every time `CC.cruiseControl.resume` went low. See `806907cd33ef2647|2023-06-02--10-24-33--2`.

Also I know the alert priority is WIP, but having it be green is a good visual cue-- it was hard to tell visually when the alert changed from "no steer" to "resume".

sshane, 2023-06-02:
Yeah I was afraid of that, besides something more complicated with hysteresis and ignoring the alert sound under certain conditions, I think we want to fix the below steering speed alert first before we can improve things around it.

nworb-cire, 2023-06-03:
One option I see is to do something like this:
```python3
if latcontrol.MIN_LATERAL_CONTROL_SPEED < ret.vEgo < self.CP.minSteerSpeed:
  events.add(EventName.belowSteerSpeed)
```
(I believe this should be equivalent to checking for `CC.latActive`, if done in a context where that is available.)

This would remove the no steer alert at standstill, and play the alert sound again when resuming. I think an argument could be made either way for this being a good or bad thing.

nworb-cire, 2023-06-14:
@sshane If the above solution is reasonable, I have a PR ready to file for it. Just let me know your thoughts.

sshane, 2023-06-14:
I think it's worth considering, but note that we have pending plans on overhauling this whole system, so we may want to wait for that - even if it will take longer.

nworb-cire, 2023-06-14:
Closing pending that.