commaai/openpilot #28365: Phantom braking event

MJR1989, 2023-06-02:
### Describe the bug

Momentary phantom braking from faulty radar speed detection of cut in vehicle 

### Provide a route where the issue occurs

b0a7a22dea55a1cd|2023-06-01--16-32-34--8

### openpilot version

0..9.2

haraschax, 2023-06-02:
[image]
[image]
Appears to be a radar issue, reports a slow car. But no slower cars visible in the video. Overpass make be causing issues.

adeebshihadeh, 2024-01-05:
@haraschax want to keep open or can we close?

closed 2024-05-17 by adeebshihadeh