commaai/openpilot #28398: "Check" button in Software tab does not translate

TheOfficialJosh, 2023-06-04:
### Describe the bug

The "Check" button does not change to match the translation file for any language.
Found while working on Spanish translation (re-open of #27844), which is why branch shows translation-es.
[image]
[image]
[image]

### Provide a route where the issue occurs

5130484aa8069bad

### openpilot version

0.9.3

adeebshihadeh, 2023-06-05:
@sshane isn't there supposed to be a test that catches this?

sshane, 2023-06-05:
Dynamically set text is not very well caught in the test, it only currently tests one static UI state, and so it never gets to these calls, which rely on the updater process. @maxime-desroches maybe as a future project we can test more states of the UI

https://github.com/commaai/openpilot/blame/6d91b7c341afd1bf38e1bb044338a2026d0a6c94/selfdrive/ui/qt/offroad/software_settings.cc#L129-L141

sshane, 2023-06-06:
Fixed by https://github.com/commaai/openpilot/pull/28417. opened an issue to add better tests

closed 2023-06-06 by sshane