commaai/openpilot #28470: comma three display init fails

VoltIcaRus, 2023-06-09:
### Describe the bug

I just received my comma3 yesterday and after checking the 
I used the adapter in the picture to turn it on, and then used the 
I disassembled it and installed the Samsung 980 SSD.
During the assembly process, I did not touch the display connector or remove it, just slide the display to install the SSD. 

It hasn't turned on since then

(I haven't even installed it in the car yet)

I got lucky once and it turned on. 
The moment I picked it up to retighten the screws, the screen turned off again. 
and it hasn't turned on since. 

I've read tons of stuff on the discord and wiki. 

I assumed that the device itself was in good working order as it was applying power and the cooling fan was still running.
For further clarification, I have attempted to do the following 

1. let it discharge for 20 minutes and then connected it using comma power without starting the car. 

2. disconnected the SSD and powered it up. 

3. reinstall the SSD

4. reinstalled the display connector

5. entered fastboot mode and reflashed agnos on windows pc.

6. installed ubuntu and tried serial attach 

 When I tried serial attach, I got the following logs 
UEFI seems to keep rebooting
resulting in 
So I couldn't even log in with comma comma.
The log keeps repeating 

I think I've tried everything I can think of through a lot of resources.
Any help would be appreciated.

I didn't even see the dongle ID 

[image]

```Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.2.0-00371-SDM845LZB-1
S - IMAGE_VARIANT_STRING=SDM845LA
S - OEM_IMAGE_VERSION_STRING=5597b19e995a
S - Boot Interface: UFS
S - Secure Boot: Off
S - Boot Config @ 0x00786070 = 0x000000c1
S - JTAG ID @ 0x00786130 = 0x2008e0e1
S - OEM ID @ 0x00786138 = 0x00000000
S - Serial Number @ 0x00784138 = 0xec65ed52
S - OEM Config Row 0 @ 0x00784188 = 0x0000000400000000
S - OEM Config Row 1 @ 0x00784190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x007841a0 = 0x0050200080000400
S - Feature Config Row 1 @ 0x007841a8 = 0xe000ffff00007fff
S - Core 0 Frequency, 1516 MHz
S - PBL Patch Ver: 7
S - PBL freq: 600 MHZ
B -        94 - PBL, Start
B -      5284 - bootable_media_detect_entry
B -     57841 - bootable_media_detect_success
B -     58218 - elf_loader_entry
B -     58829 - auth_hash_seg_entry
B -     59031 - auth_hash_seg_exit
B -     65490 - elf_segs_hash_verify_entry
[... 379 lines cut ...]
[   36.221537] dsi-hw:[DSI_1] debug bus ctrl: 0x161 status:0x0
[   36.227235] dsi-hw:[DSI_1] debug bus ctrl: 0x1161 status:0x0
[   36.233119] dsi-hw:[DSI_1] debug bus ctrl: 0x171 status:0x0
[   36.238818] dsi-hw:[DSI_1] debug bus ctrl: 0x171 status:0x0
[   36.244755] dsi-hw:[DSI_1] debug bus ctrl: 0x181 status:0x0
[   36.250536] dsi-hw:[DSI_1] debug bus ctrl: 0x181 status:0x0
[   36.256347] dsi-hw:[DSI_1] debug bus ctrl: 0x191 status:0x0
[   36.262119] dsi-hw:[DSI_1] debug bus ctrl: 0x191 status:0x0
[   36.267902] dsi-hw:[DSI_1] debug bus ctrl: 0x1a1 status:0x0
[   36.273747] dsi-hw:[DSI_1] debug bus ctrl: 0x1a1 status:0x0
[   36.279438] dsi-hw:[DSI_1] debug bus ctrl: 0x1b1 status:0x0
[   36.285261] dsi-hw:[DSI_1] debug bus ctrl: 0x1b1 status:0x0
[   36.291051] dsi-hw:[DSI_1] debug bus ctrl: 0x11b1 status:0x0

Ubuntu 20.04.5 LTS tici ttyMSM0

tici login: ```

### Provide a route where the issue occurs

nope

### openpilot version

nope

robbederks, 2023-06-09:
For the hardware issues, please contact our support through https://comma.ai/support

The touch controller not being in application mode might be a software bug though. It's supposed to be flashed and put back in application mode by the kernel driver. I'll double check that flow still works, but this should be unrelated to the display not working (only touch might not function with this error).

VoltIcaRus, 2023-06-11:
Reflashing via web seems to have been successful
I successfully logged in with a comma comma.
Is there a command that might help?

VoltIcaRus, 2023-06-11:
I tried flashing my web browser and I can now log in with comma comma comma 

But I still get 
Touch screen is not in application mode (returned 0x10). Cannot init
Failed to initialize touch: 7
Could not initialize touch: 7 I see an error 

```-----------------------------
Platform Init  [ 2463] BDS
Platform   : TICI
Chip Name  : SDA845
Chip Ver   : 2.1
Core 0 Freq: 1612 MHz
-----------------------------
UEFI Total : 1087 ms
POST Time      [ 2464] OS Loader
Loader Build Info: May 12 2023 18:35:54
comma 845 Agnos bootloader
VB: RWDeviceState: Succeed using devinfo!
Launching fastboot
Fastboot Build Info: May 12 2023 18:35:53
qusb2_1: hstx: 7
PLL1 locked: 5
usb_lane: 0
Fastboot: Initializing...
Fastboot: Processing commands
Touch screen is not in application mode (returned 0x10). Cannot init
Failed to initialize touch: 7
Could not initialize touch: 7
Maximum boot temp check passed!
No fastboot events recorded. Quitting fastboot.
Active Slot _a is bootable, retry count 1
Booting from slot (_a)
Loading Image Start : 5627 ms
Loading Image Done : 5627 ms
Total Image Read size : 4096 Bytes
Loading Image Start : 5627 ms
Loading Image Done : 5656 ms
Total Image Read size : 15155200 Bytes
Loading Image Start : 5658 ms
No dtbo partition is found, Skip dtbo
VB: DeviceInit: Device is unlocked! Skipping verification!
VB: LoadImageAndAuthForLE complete!
Exit key detection timer
UsbfnDwcTransfer: Ep 1, EpDir 1: USB cable not attached, cannot transfer data
UsbDeviceSend: Failed to submit data transfer: Not Ready
No Ffbm cookie found, ignore: Not Found
Memory Base Address: 0x80000000
Decompressing kernel image start: 5697 ms
Decompressing kernel image done: 6049 ms
Cmdline: console=ttyMSM0,115200n8 quiet loglevel=3 earlycon=msm_geni_serial,0xA84000 androidboot.hardware=qcom androidboot.console=ttyMSM0 video=DSI-1:1080x2160@60e ehci-hcd.park=3 lpm_levels.sleep_disabled=1 service_locator.enable=1 androidboot.selinux=p
BootLinux: failed to get dtbo image
Best match DTB tags 341/00000020/0x00000000/20001/20014/20115/20018/0/(offset)0x98CE4C3F/(size)0x0005B5EC
Using pmic info 0x20014/0x20115/0x20018/0x0 for device 0x20014/0x20115/0x20018/0x0
Memory Base Address: 0x80000000
RAM Partitions
Add Base: 0x0000000080000000 Available Length: 0x00000000FDFA0000 
WARNING: Could not find mem-offline node.
ERROR: Could not get splash memory region node
kaslr-Seed is added to chosen node

Shutting Down UEFI Boot Services: 6207 ms
BDS: LogFs sync skipped, Unsupported
App Log Flush : 12 ms
Exit BS        [ 6224] UEFI End
[    1.318466] pcie:pcie_init.
^[[A^[[A^[[A^[[A^[[A^[[A^[[A^[[A^[[ASELinux:  Could not open policy file <= /etc/selinux/targeted/policy/policy.32:  No such file or directory
[   29.103728] FG: fg_get_time_to_full_locked: battery profile is not loaded
[   29.443136] FG: fg_get_time_to_full_locked: battery profile is not loaded
[   30.086847] F2FS-fs (sda11): Can't find valid F2FS filesystem in 1th superblock
[... 205 lines cut ...]
                   h8888888888888k 
                   t888888888s888k 
                    `t88888d/ h88k 
                       ```    h88l 
                             ,88k` 
                            .d8h` 
                           +d8h   
                        _+d8h`     
                      ;y8h+`       
                      |-`

Welcome to AGNOS (GNU/Linux 4.9.103+ aarch64)

 System information disabled due to load higher than 8.0

The programs included with the Ubuntu system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Ubuntu comes with ABSOLUTELY NO WARRANTY, to the extent permitted by
applicable law.

comma@tici:~$ ```

Are there any commands I should try?

robbederks, 2023-06-11:
Not really, it's likely related to your hardware issue...

robbederks, 2023-06-12:
I've confirmed that the touch bootloader exit in the kernel still works as expected, so the rest of this issue is for our hardware support!

closed 2023-06-12 by robbederks