commaai/openpilot #28483: fuzzy fingerprinting: can match to a car while missing crucial ECUs

sshane, 2023-06-10:
For example, if the radar or camera refuses to respond (or any other important ECU), but we get enough matches with other ECUs, we would match when we really shouldn't.

sshane, 2023-06-10:
@jayvin100 I would recommend trying a different OBD-C cable or ethernet cable since this looks to be a hardware issue. panda drops some messages on both OBD 2 port and powertrain bus right around when we expect responses from the camera ECU.

[image]

Nothing back from ECU:

[image]

@robbederks does this look like a harness box issue or cable issue?

robbederks, 2023-06-10:
Does this work on other 2022 civics? Strange that there would be a drop on two buses

jayvin100, 2023-06-10:
> Just looking into that case since we would fail to fingerprint for that user after this change.
> 
> @jayvin100 I would recommend trying a different OBD-C cable or ethernet cable since this looks to be a hardware issue. panda drops some messages on both OBD 2 port and powertrain bus right around when we expect responses from the camera ECU.
> 
> `8fcafab4167b8c6c|2023-06-08--12-24-23`: [image]
> 
> Nothing back from ECU:
> 
> [image]
> 
> @robbederks does this look like a harness box issue or cable issue?

Battery just died, got a fresh one installed. Not sure if that could cause the issue also am on sunnypilot running nicki model, shall I install back stock openpilot to see if this issue persist?

adeebshihadeh, 2023-06-15:
Yes, make sure you can repro on stock openpilot. Do a bunch of these users have a community built harness or something?

jayvin100, 2023-06-16:
> Yes, make sure you can repro on stock openpilot. Do a bunch of these users have a community built harness or something?

Have been running master-ci since this issue was brought up last week. It's has been close to a week now, is it giving the expected responses? @sshane

adeebshihadeh, 2024-04-15:
@sshane isn't this fixed?

sshane, 2024-04-15:
this issue? no. the civic blocker? yes. i need to check some data to make sure this isn't the case for other cars and we can merge https://github.com/commaai/openpilot/pull/28481