commaai/openpilot #28484: Communication Problem Between Processes

Euclid71, 2023-06-10:
### Describe the bug

This is the same or similar to closed issue #21873. 

I get this error with the same symptoms as issue #21873 when I have the OpenStreetMap database set to "== None ==". 

When I set the database to "US - Pacific", I never get this error.

Route: f1bc09b7aead608d|2023-06-09--23-37-12--1
All logs are uploaded.

### Provide a route where the issue occurs

f1bc09b7aead608d|2023-06-09--23-37-12--1

### openpilot version

0.8.16

vanillagorillaa, 2023-06-10:
You are running a fork, dragonpilot, and not stock openpilot. If your issues persists on stock openpilot feel free to reopen this issue, otherwise you will want to report this issue to your fork maintainer on their respective repo.

closed 2023-06-10 by vanillagorillaa

lhn136, 2023-06-24:
I recently ran into the same issue with dragonpilot