commaai/openpilot #28486: service power Steering on a 2021 Ram 1500 DT Laramie ecodiesel

joecatran, 2023-06-10:
### Describe the bug

anyone want to take a crack at my dilemma using comma three and a 2021 Ram 1500 DT Laramie Eco diesel? I would appreciate it
No matter what generation 2 cable I use because I’ve tried various ones including the stock one that comes from comma three I keep getting a service power Steering error message on my dashboard. Trying to set this up in my 2021 Ram 1500 DT Laramie eco diesel. I have also attempted uninstalling and reinstalling an open pilot.
Some one also asked me these questions
When do you get the error? 
I get the error message as soon as my truck hits 30 mph which I noticed 30 mph is the speed necessary to achieve before comma 3 using open pilot to engage
What device are you using: C2, C3, custom hardware?
I'm using a comma 3 six parts came with this kit, 3, the harness for my ram pickup truck, the USB C to Ethernet cable, the generation two USB C to USB C cable, the mounts, and the OBD2 connector
Do you see the front camera video?
Yes I do see my front camera and there are videos when I've created a coma account and I see my videos uploaded
Any other messages?
Negative there are no other messages other than service cruise control on my dashboard of my 2021 Ram 1500 DT Laramie EcoDiesel and I'm assuming because this message comes on is why it's not engaging because everything seems to be working with the comma three I mean it's watching the road it's taking videos it's setting destinations if I put an address in the app it'll transmit to the device!

### Provide a route where the issue occurs

https://connect.comma.ai/06794b0d8a5d85ed/1686334412971/1686337332448

### openpilot version

0.9.2

### Additional info

I have provided a link for a route above however this occurs on a ll trips

sshane, 2023-06-26:
Hi, can you get another route with the example and upload rlogs?

joecatran, 2023-06-26:
We already figured out the problem. Iam in the process of installing ACC in my vehicle. I thought I had it already.Thanks Joe Catran 727-331-1328 Apply for Life Insurance all on your own. Scan the QR code below to apply with Ethos or CLICK HERE

closed 2023-06-26 by adeebshihadeh