commaai/openpilot #28506: Runing openpilot 0.9.13 can't show the green planning trajectory on ui

Ktter, 2023-06-12:
### Describe the bug

I tried to run openpilot 0.9.13 on Ubuntu 18.04 LTS
using some steps as follow:
1. ` ./start start_carla.sh` in ../sim in Terminal A
2. `sudo docker exec -it openpilot_client bash` in Terminal B
3. `./launch_openpilot.sh` in Terminal B
4. exec docker of openpilot again in terminal C
5. `./bridge.py --dual_camera`
[image]

I could manually control this car in _ui  two week ago, but i cant run it again now.
I will very apperaciate you if you can give me some he

### OS Version

Ubuntu 18.04 LTS

### openpilot version or commit

openpilot 0.9.13

haraschax, 2023-06-14:
There's a couple bugs with the simulator unfortunately, fixing them here:
https://github.com/commaai/openpilot/pull/28541

This branch engages in simulation, need to clean it up before I can merge though

haraschax, 2023-06-14:
Fixed

closed 2023-06-14 by haraschax