commaai/openpilot #28518: hyundai openpilot long jerky accel/decel behavior at stop

jacyhung, 2023-06-13:
### Describe the bug

Not sure if this is a model issue, or a Hyundai op long issue (did not experience this on older 0.8 releases of OP with hyundai op long). This behavior occurs a lot when behind a lead car, or when stopped at a red light, on both experimental mode and regular op long. The car will come to a stop (sometimes a little too far from the lead), then in rapid succession accel/decel and repeat, resulting in a jerky movement & requiring disengagement. The route provided is in experimental mode, and the behavior is shown at both a red light and with a lead. Was testing the nav branch, but this issue occurs pretty much on any master-based branch that allows for hyundai op long.

Stopped with lead car - 03:18:08
Stopped at red light - 09:24:02

### Which car does this affect?

2021 Hyundai Sonata

### Provide a route where the issue occurs

b0af2fff166d80fe|2023-06-12--09-14-42

### openpilot version

0.9.3 - f0c1e000c6ef09e61232e6afb4c3a658035afbb9

sunnyhaibin, 2023-06-13:
I can second this. This has been a known issue on my 2021 Hyundai Sonata N Line since 0.8.9 (when openpilot longitudinal was first supported on HKG CAN platform.

thealexbaron, 2023-06-26:
Copied from duplicate ticket that I just closed:

### Describe the bug

I am using a Comma 3 with Open Pilot on a 2020 Hyundai Sonata using Experimental mode. This issue seems to happen regardless of the situation as long as my car has been brought to a complete stop with Open Pilot engaged. As you can see in the video, my car just continually inches forward even after coming to a complete stop and the car in front of me has not moved at all. It also sometimes will do a rapid pumping of the brakes which makes a really weird sound.

The issue is very clearly happening at 18:27:58 within the route mentioned in this issue 

### Which car does this affect?

Hyundai Sonota 2020

### Provide a route where the issue occurs

e8b1f9ef35c46678|2023-06-23--18-11-12--16

### openpilot version

0.9.4

### Additional info

_No response_

thealexbaron, 2023-08-23:
Are there any updates on this? It's becoming pretty painful to use e2e with this issue. Is it possible that I can contribute to raise the priority of this ticket? 

I love my comma/open pilot. I use it constantly. Any help is appreciated!

jakethesnake420, 2023-09-02:
Some cars require a hold signal to be toggled. You can compare the stock system signals when coming to a stop and Openpilots emulation. There could be another signal that represents hold/release.

alecjsullivan, 2023-09-04:
Radar tracks enabled, 22 sonata, OP Long 0.9.4 Maybe related to tracks? Seems like the cars is getting accelerate and brake commands simultaneously. Brakes will scrub during issue.