commaai/openpilot #28544: The openpilot simulation can be run, but the ui can't always display the image.

Ktter, 2023-06-14:
### Describe the bug

I can run the sim using command as follow:
1.` ./start_carla.sh`
2.`./start_openpilot_docker.sh`
then the ui has no any change but the panda state

I would be very grateful if you could give me some help!
[image]

### OS Version

Ubuntu18.04

### openpilot version or commit

../openpilot_sim:latest

haraschax, 2023-06-14:
Not sure what issue you are having. I just tried this and it worked on lastest master for me. It does say bridge is lagging, that isn't normal. Do you have openpilot compiled and running on your PC? If so I would just try running ./launch_openpilot.sh in tools/sim and bridge, instead of the docker. Does that work?

haraschax, 2023-06-18:
Closing, need more info to reopen

closed 2023-06-18 by haraschax