commaai/openpilot #28560: controlsd crash on fingerprint miss

jyoung8607, 2023-06-15:
### Describe the bug

A new openpilot user installed master for their first driving experience on a 2017 Golf, and reported "openpilot unavailable / waiting for controls to start". Upon connecting via SSH, found this in tmux:

[image]

It's not unusual for a VW to need a fingerprint update, but it should be last-resort fingerprinting as "mock" for which a stub openpilot port exists, not as "debug" which doesn't exist. Not sure where this is coming from.

### Provide a route where the issue occurs

cd154701ab39149d|2023-06-15--15-14-15

### openpilot version

e57f3ab4daeb18ac1c8786b3b5c7d0ad7b000646

### Additional info

Was able to collect fingerprint info normally after I had the user install current release3.

adeebshihadeh, 2023-06-15:
@sshane it's fuzzy fingerprinting. we need better tests to catch this

jyoung8607, 2023-06-15:
Agreed this may be a general breakage of VW fuzzy fingerprinting. I just had it pop up with another user. He was running outdated master (never got off his temp fingerprinting branch) and it's gotten an ECU flash update since.

Last working route, fuzzyFingerprint=True: `b379b0c2f98a5838|2023-06-15--18-08-58`
Failed attempt to run a test branch derived from today's master: `b379b0c2f98a5838|2023-06-15--20-40-19`

closed 2023-06-15 by sshane