commaai/openpilot #28582: Ford: "Park not available" alert

sshane, 2023-06-17:
One of the Nissan queries puts the Bronco Sport into a neutral tow mode state, but it doesn't have neutral tow mode...

sshane, 2023-06-17:
Here you can see everything was normal on startup where I was in drive, then we queried with this Nissan request and the various neutral tow signals activated, and the normally "OutOfPark
`

[image]

sshane, 2023-06-17:
Random Nissan X-Trail FW query:

fa9224be239005c7|2023-05-30--08-49-15
[image]

adeebshihadeh, 2023-07-09:
@sshane is this resolved?

sshane, 2023-07-09:
No, we still need to merge the Nissan diagnostic mode change.

incognitojam, 2023-07-16:
Is there anything we can do to fix this in the meantime, perhaps sending Nissan queries after Ford queries? It does cause an annoyance for users (reported that the car will always try to activate the parking brake when you slow down).

sshane, 2023-07-16:
The problem occurs when the Ford fuzzy fingerprints (since we run all queries before fuzzing fingerprinting, can be improved), or doesn't fingerprint at all. Ordering doesn't matter at the moment.

closed 2023-07-18 by sshane