commaai/openpilot #28603: Building source code occured errors!!

Ktter, 2023-06-20:
### Describe the bug

I tried to build the openpilot according `https://github.com/commaai/openpilot/blob/master/tools/README.md`. But some errors occured when i executed the command `scons -u -j$(nproc)` .
more information as follows:
`/usr/bin/ld: tools/replay/libqt_replay.a(consoleui.o): undefined reference to symbol 'keypad'
/usr/bin/ld: /home/wx/anaconda3/lib/./libtinfow.so.6: error adding symbols: DSO missing from command line
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [tools/replay/replay] Error 1
scons: building terminated because of errors.`

### OS Version

Ubuntu 20.04 LTS

deanlee, 2023-06-20:
Make sure that you have installed `libncurses5-dev` and `libncursesw5-dev` \ correctly and the tinfo is included in pkg config:
```

@: ncursesw5-config --libs
-Wl,-Bsymbolic-functions -lncursesw -ltinfo
```

Ktter, 2023-06-21:
> Make sure that you have installed `libncurses5-dev` and `libncursesw5-dev` \ correctly and the tinfo is included in pkg config:
> 
> ```
> 
> @: ncursesw5-config --libs
> -Wl,-Bsymbolic-functions -lncursesw -ltinfo
> ```

Thank you, i have installed `libncurses5-dev` and `libncursesw5-dev`. I execute command `ncursesw5-config --libs` and the terminal can show me the same `infomations:`-Wl,-Bsymbolic-functions -lncursesw -ltinfo`. But i still can't build the op successfully.