commaai/openpilot #28644: Communication Issue Between Processes upon engaging openpilot shortly after cold boot

goodchoom, 2023-06-22:
### Describe the bug

openpilot threw a Communication Issue Between Processes alert when I engaged openpilot shortly after going onroad from a cold boot, and as soon as the openpilot logo was visible on-screen. I usually don't engage openpilot this quickly, but this time I opted to do so because I had already been driving for quite some time waiting several minutes for the device to boot up, as it usually does after going an entire day without turning on my car.

### Provide a route where the issue occurs

4f3e4f2f6d94e940|2023-06-22--07-11-14--1

### openpilot version

eb9a902010dc5c3a10bbd1addace15f9f24d4495

adeebshihadeh, 2023-06-22:
`lateralPlan` went invalid for a few frames, but it's not immediately clear why. All of the services it listens to were fine and `longitudinalPlan` was also fine.

adeebshihadeh, 2023-06-23:
This turned out to be fairly subtle. Startup normally handles issues like this with an [initializing period](https://github.com/commaai/openpilot/blob/7d6c22949a42b90fa2fc67608a53f6cdce52ebac/selfdrive/controls/controlsd.py#L436) in controlsd that waits until all services are valid. 

In this case, there was a single frame of modelV2 on startup that caused plannerd's SubMaster to go invalid once all the [zeros](https://github.com/commaai/cereal/blob/0a1d3c4c9aaaac9567052d84e7ccac56a4673b70/messaging/__init__.py#L169) in `sm.recv_dts` were out. https://github.com/commaai/cereal/pull/487 handles this.

closed 2023-06-23 by adeebshihadeh