commaai/openpilot #28651: LEXUS NX-300h hybrid 2020  [vehicle is not recognized and is in dashcam mode]

Abanggo, 2023-06-23:
### Describe the bug

The vehicle is not recognized and is in dashcam mode.

     NX Hybrid 2020, (comma.ai homepage) 
     openpilot upgrades your Lexus NX Hybrid with automated lane centering at all speeds, and adaptive cruise control that automatically resumes from a stop. Traffic light and stop sign handling is also available in Experimental mode.
     SUPPORT: openpilot will work with all packages and trims of this car.

You must be busy, but please help. thank you.

[image]

### Provide a route where the issue occurs

 9a1f34e34fb917f2 | 2023-06-23--11-25-02--0

### openpilot version

master-ci v0.9.4 release  7d6c229

### Additional info

[image]
[image]

Abanggo, 2023-06-24:
Thank you very much for your interest.

sshane, 2023-06-24:
Thanks for trying master with my addition of your hybrid ECU, we got it! This will help us improve fingerprinting in the near future for Toyota.

```python
{'address': 2018,
                          'brand': 'toyota',
                          'bus': 0,
                          'ecu': 'hybrid',
                          'fwVersion': b'\x05899837851000\x00\x00\x00\x00899857801000\x00\x00\x00\x00898844813300\x00\x00\x00\x0089884481430'
                                       b'0\x00\x00\x00\x00898844815300\x00\x00\x00\x00',
                          'logging': True,
                          'obdMultiplexing': True,
                          'request': [b'>', b'\t\x04'],
                          'responseAddress': 2026,
                          'subAddress': 0},
```

Added your FW, can you try this branch? https://github.com/commaai/openpilot/tree/add-missing-lexus-300h-fw

Installer URL: https://installer.comma.ai/commaai/add-missing-lexus-300h-fw

Abanggo, 2023-06-24:
Dear Sshane
Thank you very much for your attention and solving it.
Now, let's install and use it on the C3 of the vehicle.
If you have any other opinions, please leave a comment here.
thank you

sshane, 2023-06-24:
You can type that installer URL in after factory resetting your device and it will install.

Abanggo, 2023-06-24:
thank you

The result of installing the latest version and operating it with the vehicle confirmed that it works well.

Is it correct to reboot again by clicking on just one power off?

thank you so much. I will use it well.

[image]

[image]

Abanggo, 2023-06-26:
Thanks for helping us use openpilot.
I've been driving around town and it's working very well.
Thank you very much for your help.

closed 2023-06-26 by sshane

Abanggo, 2023-06-28:
**[Report Abnormal Behavior]**

LEXUS NX-300h hybrid 2020 [vehicle is not recognized and is in dashcam mode] #28651
Lexus: add missing NX 300H 2021 FW #28678

Perform a factory reset and [installer.comma.ai/commaai/add-missing-lexus-300h-fw]
Installed on June 26, 2023 and drove about 400 km in normal operation and parked.

June 27, 2023 [dmonitoringmodeld, navmodeld] is displayed when the vehicle is started, and normal operation stops.

On June 28, 2023, the master / master-ci version was installed and a test drive was performed, but the same [dmonitoringmodeld, navmodeld] was displayed and did not operate normally.

Translated through Google Translate.

[image]

Abanggo, 2023-06-29:
https://github.com/commaai/openpilot/issues/28726