commaai/openpilot #28681: Invalid PEP 440 version: '0.23ubuntu1' during install on ubuntu 20.04

Narolth, 2023-06-24:
### Describe the bug

Fresh install of Ubuntu 20.04, following https://github.com/commaai/openpilot/blob/master/tools/README.md Step for 
# for Ubuntu 20.04 LTS 
tools/ubuntu_setup.sh 
results in :

`update pip
Requirement already satisfied: pip==23.1.2 in /home/administrator/.pyenv/versions/3.8.10/lib/python3.8/site-packages (23.1.2)
Requirement already satisfied: poetry==1.2.2 in /home/administrator/.local/lib/python3.8/site-packages (1.2.2)
Requirement already satisfied: cachecontrol[filecache]<0.13.0,>=0.12.9 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (0.12.14)
Requirement already satisfied: cachy<0.4.0,>=0.3.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (0.3.0)
Requirement already satisfied: cleo<2.0.0,>=1.0.0a5 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.0.0a5)
Requirement already satisfied: crashtest<0.4.0,>=0.3.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (0.3.1)
Requirement already satisfied: dulwich<0.21.0,>=0.20.46 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (0.20.50)
Requirement already satisfied: html5lib<2.0,>=1.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.1)
Requirement already satisfied: importlib-metadata<5.0,>=4.4 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (4.13.0)
Requirement already satisfied: jsonschema<5.0.0,>=4.10.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (4.17.3)
Requirement already satisfied: keyring>=21.2.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (24.1.0)
Requirement already satisfied: packaging>=20.4 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (23.1)
Requirement already satisfied: pexpect<5.0.0,>=4.7.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (4.8.0)
Requirement already satisfied: pkginfo<2.0,>=1.5 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.9.6)
Requirement already satisfied: platformdirs<3.0.0,>=2.5.2 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (2.6.2)
Requirement already satisfied: poetry-core==1.3.2 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.3.2)
Requirement already satisfied: poetry-plugin-export<2.0.0,>=1.1.2 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.2.0)
Requirement already satisfied: requests<3.0,>=2.18 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (2.28.1)
Requirement already satisfied: requests-toolbelt<0.10.0,>=0.9.1 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (0.9.1)
Requirement already satisfied: shellingham<2.0,>=1.5 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.5.0.post1)
Requirement already satisfied: tomlkit!=0.11.2,!=0.11.3,<1.0.0,>=0.11.1 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (0.11.8)
Requirement already satisfied: urllib3<2.0.0,>=1.26.0 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (1.26.16)
Requirement already satisfied: virtualenv!=20.4.5,!=20.4.6,>=20.4.3 in /home/administrator/.local/lib/python3.8/site-packages (from poetry==1.2.2) (20.21.1)
Requirement already satisfied: msgpack>=0.5.2 in /home/administrator/.local/lib/python3.8/site-packages (from cachecontrol[filecache]<0.13.0,>=0.12.9->poetry==1.2.2) (1.0.5)
Requirement already satisfied: lockfile>=0.9 in /home/administrator/.local/lib/python3.8/site-packages (from cachecontrol[filecache]<0.13.0,>=0.12.9->poetry==1.2.2) (0.12.2)
Requirement already satisfied: pylev<2.0.0,>=1.3.0 in /home/administrator/.local/lib/python3.8/site-packages (from cleo<2.0.0,>=1.0.0a5->poetry==1.2.2) (1.4.0)
Requirement already satisfied: six>=1.9 in /home/administrator/.pyenv/versions/3.8.10/lib/python3.8/site-packages (from html5lib<2.0,>=1.0->poetry==1.2.2) (1.16.0)
Requirement already satisfied: webencodings in /home/administrator/.local/lib/python3.8/site-packages (from html5lib<2.0,>=1.0->poetry==1.2.2) (0.5.1)
Requirement already satisfied: zipp>=0.5 in /home/administrator/.local/lib/python3.8/site-packages (from importlib-metadata<5.0,>=4.4->poetry==1.2.2) (3.15.0)
Requirement already satisfied: attrs>=17.4.0 in /home/administrator/.local/lib/python3.8/site-packages (from jsonschema<5.0.0,>=4.10.0->poetry==1.2.2) (23.1.0)
Requirement already satisfied: importlib-resources>=1.4.0 in /home/administrator/.local/lib/python3.8/site-packages (from jsonschema<5.0.0,>=4.10.0->poetry==1.2.2) (5.12.0)
Requirement already satisfied: pkgutil-resolve-name>=1.3.10 in /home/administrator/.local/lib/python3.8/site-packages (from jsonschema<5.0.0,>=4.10.0->poetry==1.2.2) (1.3.10)
Requirement already satisfied: pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 in /home/administrator/.local/lib/python3.8/site-packages (from jsonschema<5.0.0,>=4.10.0->poetry==1.2.2) (0.19.3)
Requirement already satisfied: jaraco.classes in /home/administrator/.local/lib/python3.8/site-packages (from keyring>=21.2.0->poetry==1.2.2) (3.2.3)
Requirement already satisfied: SecretStorage>=3.2 in /home/administrator/.local/lib/python3.8/site-packages (from keyring>=21.2.0->poetry==1.2.2) (3.3.3)
Requirement already satisfied: jeepney>=0.4.2 in /home/administrator/.local/lib/python3.8/site-packages (from keyring>=21.2.0->poetry==1.2.2) (0.8.0)
Requirement already satisfied: ptyprocess>=0.5 in /home/administrator/.local/lib/python3.8/site-packages (from pexpect<5.0.0,>=4.7.0->poetry==1.2.2) (0.7.0)
Requirement already satisfied: charset-normalizer<3,>=2 in /home/administrator/.local/lib/python3.8/site-packages (from requests<3.0,>=2.18->poetry==1.2.2) (2.1.1)
Requirement already satisfied: idna<4,>=2.5 in /home/administrator/.pyenv/versions/3.8.10/lib/python3.8/site-packages (from requests<3.0,>=2.18->poetry==1.2.2) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in /home/administrator/.local/lib/python3.8/site-packages (from requests<3.0,>=2.18->poetry==1.2.2) (2022.12.7)
Requirement already satisfied: distlib<1,>=0.3.6 in /home/administrator/.local/lib/python3.8/site-packages (from virtualenv!=20.4.5,!=20.4.6,>=20.4.3->poetry==1.2.2) (0.3.6)
Requirement already satisfied: filelock<4,>=3.4.1 in /home/administrator/.local/lib/python3.8/site-packages (from virtualenv!=20.4.5,!=20.4.6,>=20.4.3->poetry==1.2.2) (3.12.2)
Requirement already satisfied: cryptography>=2.0 in /home/administrator/.pyenv/versions/3.8.10/lib/python3.8/site-packages (from SecretStorage>=3.2->keyring>=21.2.0->poetry==1.2.2) (41.0.1)
Requirement already satisfied: more-itertools in /home/administrator/.pyenv/versions/3.8.10/lib/python3.8/site-packages (from jaraco.classes->keyring>=21.2.0->poetry==1.2.2) (9.1.0)
Requirement already satisfied: cffi>=1.12 in /home/administrator/.local/lib/python3.8/site-packages (from cryptography>=2.0->SecretStorage>=3.2->keyring>=21.2.0->poetry==1.2.2) (1.15.1)
Requirement already satisfied: pycparser in /home/administrator/.local/lib/python3.8/site-packages (from cffi>=1.12->cryptography>=2.0->SecretStorage>=3.2->keyring>=21.2.0->poetry==1.2.2) (2.21)

Invalid PEP 440 version: '0.23ubuntu1'`

### OS Version

Ubuntu 20.04.6 LTS

### openpilot version or commit

cat common/version.h #define COMMA_VERSION "0.9.4"

adeebshihadeh, 2023-06-25:
Are you installing in a virtualenv? Looks like this can be caused by conflicts with the system python.

Narolth, 2023-06-25:
> 

Nope, it's running on physical hardware. Specifically a brand new Ubuntu install on a Microsoft Surface Pro 6.

adeebshihadeh, 2023-07-01:
A virtualenv isn't related to your hardware; see https://docs.python.org/3/library/venv.html. 

Regardless, this isn't reproducible locally on my machine or in a clean environment in CI. It seems related to some state on your machine. Moving to a discussion.