commaai/openpilot #28682: Toyota Corolla Hybrid: Extremely slow acceleration and early braking when using 'Aggressive' mode

TKassis, 2023-06-24:
### Describe the bug

Aggressive mode seems to be opposite of what it intends to do. Acceleration when the lead starts moving is extremely slow and the car tends to brake way too early when stopping for a lead. I've always had these issues with OpenPilot, but though 'Aggressive' mode would help address them, but for some reason it makes them substantially worse.

### Which car does this affect?

Toyota Corolla Hybrid 2021

### Provide a route where the issue occurs

ecebb0b63434e2f2|2023-06-23--12-14-31--6

### openpilot version

0.9.4

sshane, 2023-06-26:
Can you get a new route with rlogs? The ones for this route are gone already. It looks like (from the rough qlogs) when coming to a stop, it didn't overshoot braking too badly, and was generally following the plan well, so it doesn't look like a control issue:

[image]

But it did lag when resuming later in the segment, I looking into how the PCM transforms our acceleration request when I have free time here and there.

haraschax, 2023-06-27:
The brake response doesn't look great, but I don't think it's unexpected.

sshane, 2023-08-08:
@TKassis do you have any more recent examples?

TKassis, 2023-08-08:
Not right now. My Comma is on it's way back to me after some hardware repairs, but I'll upload some more drive when I get it later this week.

TKassis, 2023-08-25:
Here are a few more segments with extremely slow acceleration (some in aggressive and some in standard although these settings don't seem to affect my acceleration):

b434e73e06865d48|2023-08-23--17-53-56--18

b434e73e06865d48|2023-08-23--17-53-56--23

b434e73e06865d48|2023-08-23--17-53-56--23

b434e73e06865d48|2023-08-23--17-53-56--31