commaai/openpilot #28726: "Process Not Running" - occurs [dmonitoringmodeld, navmodeld]

Abanggo, 2023-06-28:
### Describe the bug

1. After github action regarding the problem of entering "Dashcam Mode"
    - Install installer.comma.ai/commaai/add-missing-lexus-300h-fw and confirm operation as a result of the test
    - Driving about 400 km on June 26 in normal operation
    - On June 27, suddenly "Process Not Running" - [dmonitoringmodeld, navmodeld] occurred

2. 6.28 c3 factory reset - master installation - driving
   - "Process Not Running" - occurs [dmonitoringmodeld, navmodeld]
     master        9a1f34e34fb917f2|2023-06-28--13-33-10--0

3. 6.28 c3 factory reset - install master-ci - drive
   - "Process Not Running" - occurs [dmonitoringmodeld, navmodeld]
     master-ci    9a1f34e34fb917f2|2023-06-28--15-06-57--0

4. Occurrence of WiFi recognition intermittently

use google translator

[image]

### Which car does this affect?

Lexus NX-300h Hybrid 2020

### Provide a route where the issue occurs

9a1f34e34fb917f2|2023-06-28--13-33-10--0

### openpilot version

master / master-ci

adeebshihadeh, 2023-06-29:
Can you connect to wifi let the full logs upload?

Abanggo, 2023-06-29:
thank you

Can you explain me how to connect to wifi and upload full logs?

Then we will work and do it according to the procedure immediately.

This was translated using Google Translate

Abanggo, 2023-06-29:
thank you

Can you explain me how to connect to wifi and upload full logs?

Then we will work and do it according to the procedure immediately.

https://useradmin.comma.ai/?onebox=9a1f34e34fb917f2

This was translated using Google Translate

{'initData': {'commands': {'entries': [{'key': 'df -h',
                                        'value': b'Filesystem      Size  Used Avail Use% Mounted on\n/dev/root       9.8G  3.7G  5.7G  40% /\ndevtmpf'
                                                 b's        1.7G     0  1.7G   0% /dev\ntmpfs           1.8G  1.1M  1.8G   1% /dev/shm\ntmpfs        '
                                                 b'   358M   18M  341M   5% /run\ntmpfs           5.0M  4.0K  5.0M   1% /run/lock\ntmpfs           1.'
                                                 b'8G     0  1.8G   0% /sys/fs/cgroup\ntmpfs           150M  284K  150M   1% /tmp\ntmpfs           12'
                                                 b'8M   11M  118M   9% /var\n/dev/sda2        27M   24K   25M   1% /persist\n/dev/sda10       12M   2'
                                                 b'4K   11M   1% /systemrw\noverlay         150M  284K  150M   1% /home\n/dev/sda12       30G  2.7G  '
                                                 b' 26G  10% /data\n/dev/nvme0n1    229G   17G  201G   8% /data/media\n/dev/sde9        28M   20M  7.'
                                                 b'3M  73% /dsp\n/dev/sde4       120M   43M   77M  36% /firmware\nnone             30G  2.7G   26G  1'
    

                                             b'0% /data/safe_staging/merged\ntmpfs           358M     0  358M   0% /run/user/0\n'},
                                       {'key': '/BUILD', 'value': b'f4b552faa665dc277c18fc0768848fda79e7b85e\n2023-05-12T12:17:44'},
                                       {'key': 'abl_a', 'value': b'2147de9402a2457bb353853ab66e1d3a847a50c18fbb2eca2b99790c5e1b8351'},
                                       {'key': 'abl_b', 'value': b'2147de9402a2457bb353853ab66e1d3a847a50c18fbb2eca2b99790c5e1b8351'},
                                       {'key': 'aop_a', 'value': b'34976ce0c97cffbf54d2326bf7c0c1724993f891a17b1c89c8f5daf7721798d3'},
                                       {'key': 'aop_b', 'value': b'34976ce0c97cffbf54d2326bf7c0c1724993f891a17b1c89c8f5daf7721798d3'},
                                       {'key': 'boot slot', 'value': b'_b'}, {'key': 'boot temp', 'value': b'BOOT_TEMP:55.6C'},
                                       {'key': 'devcfg_a', 'value': b'5e01169e3cfbce58ac8188ee21a9d0c709310e1d445bb71e77c684692141a973'},
                                       {'key': 'devcfg_b', 'value': b'5e01169e3cfbce58ac8188ee21a9d0c709310e1d445bb71e77c684692141a973'},
                                       {'key': 'lsblk',
                                        'value': b'NAME      SIZE STATE   VENDOR   MODEL                  REV SERIAL\nsda        54G running SKhynix'
                                                 b'  H28U74301AMR          D003 SKhynixH28U74301AMR_040013109BCB\n\xe2\x94\x9c\xe2\x94\x80sda1    '
                                                 b'  8K                                             \n\xe2\x94\x9c\xe2\x94\x80sda2     32M        '
                                                 b'                                     \n\xe2\x94\x9c\xe2\x94\x80sda3      1M                    '
                                                 b'                         \n\xe2\x94\x9c\xe2\x94\x80sda4    512K                                '
                                                 b'             \n\xe2\x94\x9c\xe2\x94\x80sda5    512K                                            '
                                                 b' \n\xe2\x94\x9c\xe2\x94\x80sda6   11.6G                                             \n\xe2\x94'
                                                 b'\x9c\xe2\x94\x80sda7   11.6G                                             \n\xe2\x94'
                                                 b'\x9c\xe2\x94\x80sda8    128M                                             \n\xe2\x94'
                                                 b'\x9c\xe2\x94\x80sda9     32M                                             \n\xe2\x94'
                                                 b'\x9c\xe2\x94\x80sda10    16M                                             \n\xe2\x94'
                                                 b'\x9c\xe2\x94\x80sda11   128M                                             \n\xe2\x94'
                                                 b'\x94\xe2\x94\x80sda12  30.5G                                             \nsdb         4M running'
                                                 b' SKhynix  H28U74301AMR          D003 SKhynixH28U74301AMR_040013109BCB\n\xe2\x94\x9c\xe2\x94\x80'
                                                 b'sdb1    3.5M                                             \n\xe2\x94\x94\xe2\x94\x80sdb2    356K'
                                                 b'                                             \nsdc         4M running SKhynix  H28U74301AMR      '
                                                 b'    D003 SKhynixH28U74301AMR_040013109BCB\n\xe2\x94\x9c\xe2\x94\x80sdc1    3.5M                '
                                                 b'                             \n\xe2\x94\x94\xe2\x94\x80sdc2    356K                            '
[... cut ...]
                                {'exitCode': 0, 'name': 'thermald', 'pid': 51495, 'running': True, 'shouldBeRunning': True},
                                {'exitCode': 0, 'name': 'tombstoned', 'pid': 51497, 'running': True, 'shouldBeRunning': True},
                                {'exitCode': 0, 'name': 'updated', 'pid': 0, 'running': False, 'shouldBeRunning': False},
                                {'exitCode': 0, 'name': 'uploader', 'pid': 51500, 'running': True, 'shouldBeRunning': True},
                                {

sunnyhaibin, 2023-06-29:
> thank you Can you explain me how to connect to wifi and upload full logs? Then we will work and do it according to the procedure immediately. https://useradmin.comma.ai/?onebox=9a1f34e34fb917f2 This was translated using Google Translate 2023년 6월 29일 (목) 오후 1:08, Adeeb Shihadeh ***@***.***>님이 작성:

Go to https://connect.comma.ai/, select the route with the issue, Files, find "All logs" -> click "upload x files", then "View upload queue", and confirm all raw logs are uploaded.

Full log files should not be posted here directly.

adeebshihadeh, 2023-06-29:
This started happening on 6/27 and this device otherwise worked fine since Nov 2022. I checked one of your bootlogs and it's failing to load the cdsp firmware, but other subsystems can load their firmware just fine. Did you do anything else to the device or run any other forks?

Abanggo, 2023-06-29:
thank you

I tried to install openpilot.comma.ai for the first time around the end of November 2022, but it failed.

The cause was a problem that was connected in Dashcam Mode and could not be executed.
   *** LEXUS NX-300h hybrid 2020 [vehicle is not recognized and is in dashcam mode] #28651

So, sunnypilot / drgonpilot fork was installed and tested, but dragonpilot encountered an unusual error, so the test was stopped and factory reset was performed.

After that, I installed and continued to use sunnypilot, and now I posted this problem on sunnypilot discord, but it was not resolved.

I asked the comma ai support team for help, but received a reply that analysis was not possible using sunnypilot.

After that, I knew that I could just post the problem on github, so I installed openpilot and notified the occurrence of Dashcam Mode, and it was resolved. I received a trial version and tested it. As a result, it was automatically recognized and used normally.
*** https://github.com/commaai/openpilot/tree/add-missing-lexus-300h-fw

On June 27, the problem occurred, so I posted it on github, then tested sunnypilot again and posted that the same problem was occurring in sunnypilot discode.

Nothing else has been done with the device.
I have reinstalled argos by accessing https://flash.comma.ai/.

Here's an explanation of everything I've done so far using comma three.

thank you

I used google translator.

Abanggo, 2023-06-29:
> 
> 로 이동하여 [https://connect.comma.ai/ ](https://connect.comma.ai/) 문제가 있는 경로인 파일을 선택하고 "모든 로그"를 찾은 다음 "x 파일 업로드"를 클릭한 다음 "업로드 대기열 보기"를 클릭하고 모든 원시 로그가 업로드되었는지 확인합니다. .
> 
> 전체 로그 파일을 여기에 직접 게시하면 안 됩니다.

I apologize for not knowing how to use various things.

adeebshihadeh, 2023-07-03:
Along with the user from https://github.com/commaai/openpilot/issues/28786, there are now two others hitting this. I was able to reproduce it, so I should have a branch with a fix to try soon.

Abanggo, 2023-07-03:
> #28786 의 사용자와 함께 이제 이 문제를 해결하는 다른 두 명의 사용자가 있습니다. 나는 그것을 재현할 수 있었으므로 곧 시도할 수정이 있는 분기가 있어야 합니다.

thank you
We will proceed immediately and report the results as instructed.
Thank you very much for going through the process of finding a solution.

Translated by google translator.

maweiyuweiwei, 2023-07-06:
This is the 9008 firmware package：https://drive.google.com/file/d/1YKoBl77r_kSkqlJRyZsrdKZYP7Xgdm5z/view?usp=drive_link

This is the usage method:https://mr-one.cn/2023/06/16/c3-clone-system-recovery9008-modec3%e6%95%91%e7%a0%96%e6%95%99%e7%a8%8b/

By following this action, we can fix this issue and hope it will be helpful to you，Welcome to contact me through Discord, my name is Mr. one

maweiyuweiwei, 2023-07-06:
This fault is related to the UFS partition

Abanggo, 2023-07-06:
Thank you so much for your help.
I'm a complete computer newbie, so I'm having trouble using the method you
gave me.
However, we will notify you of the related information on github #28726 to
request corrections and supplements.
thank you

adeebshihadeh, 2023-07-06:
> thank you We will proceed immediately and report the results as instructed. Thank you very much for going through the process of finding a solution.
> 
> Translated by google translator.

Please do not try this if you have an official device. This will reflash your device at a low level with unofficial firmware, and you may have to send the device in to get tixed. A software fix is coming soon with an openpilot update.

Abanggo, 2023-07-06:
I am always grateful to the team for helping drivers drive safely.
As you informed, when the official firmware is provided, we will install it
and report related matters.
thank you

ErichMoraga, 2023-07-06:
@adeebshihadeh Here's an additional user (5th incident?)... `@joeyv#5435` in Discord...
https://discord.com/channels/469524606043160576/871838269405556736/1126524680875036702

Dongle ID d084ea2309575a71
C3 s/n 254de4f7

hellokitty-666, 2023-07-06:
Unofficial solution：https://mr-one.cn/2023/06/16/c3-clone-system-recovery9008-modec3%e6%95%91%e7%a0%96%e6%95%99%e7%a8%8b/

Firmware Package：https://drive.google.com/file/d/1s3ylc3lbbiAUPrjVVCSouzd1B34p671F/view?usp=drive_link

anasufovic, 2023-07-07:
This issue also happened to me a few weeks ago, https://discord.com/channels/469524606043160576/524592892627517450/1122283823707717672 my comment there probably falsely attributes it to a mod I did. 

Hard resetting didn't resolve the issue, I tried flash.comma.ai twice and even though the flash showed as successful on the site it didn't actually work. Eventually reflashing agnos using the github repo CLI tool brought everything back into a good state.

adeebshihadeh, 2023-07-08:
Seems this was caused by flash.comma.ai erasing the `cache` partition instead of formatting it like commaai/agnos does; fixed that in https://github.com/commaai/flash/pull/3. I'll have a new AGNOS build soon that auto-formats the cache partition.

Abanggo, 2023-07-08:
Thank you for your interest and efforts to solve the problem.
We will immediately implement and test the improvements with the new method.
We will report the results.
thank you so much.

adeebshihadeh, 2023-07-09:
Can you try this branch https://github.com/commaai/openpilot/pull/28843? Uninstall openpilot in the settings and install with installer.comma.ai/commaai/agnos8.

Abanggo, 2023-07-09:
thank you so much.

installer.comma.ai/commaai/agnos8 This is the result of testing.

o Work start: 15:32-16:02 (UTC+9)
- C3 factory reset
- Wifi not recognized (problem that has occurred repeatedly so far)
    + Wi-Fi connection when restarting after about 1 minute after power off
- installer.comma.ai/commaai/agnos8

-Installing system
   + Update failed - reboot

-Installing system
   + Update failed - reboot

-Installing system
   + Update failed - reboot

After sending the email, we will proceed with factory reset and try again.

Translated by google translator.

Abanggo, 2023-07-09:
Repeatedly, but the same update failed, reboot
Conclusion failed to apply agnos8 branch.
thank you

Abanggo, 2023-07-09:
agnos8 branch rework content 2023.7.9 sun 16:15~16:47 (UTC +9)
1. https://flash.comma.ai/
     - flash.comma.ai version: ed03d96
       * Carry out work as described instructions
      
2. installer.comma.ai/commaai/agnos8
    - Wifi not recognized (problem that has occurred repeatedly so far)
       + Measure to enable access to 3 Wifi and install
          * Wi-Fi access when restarting after about 1 minute after power off
   -Installing system
   - Install adnos8
      + openpilot 0.9.4/agnos8/38ccb5b/Jul 09
   - Scheduled to report the results after testing on the vehicle

thank you so much.

Abanggo, 2023-07-09:
As a result of a test drive of about 6 km in the vehicle, I confirmed that it works well.
thank you.

adeebshihadeh, 2023-07-09:
Great. Still having WiFi issues on the AGNOS 8 branch?

Abanggo, 2023-07-10:
After agnos 8.01 is installed, Wi-Fi recognition is fine.

Reboot: Good

Power off: Good

PR #28843 (2023-07-21): agnos 8
* shaved 2s more off boot time
* `pip install` works
* python 3.11.4
* new xbl/abl
* closes #28522
* closes #28726

## release checklist

- [x] openpilot CPU usage unchanged
- [x] wifi
- [x] modem
- [x] image size
- [x] sounds
- [x] `python` is our python, not system version
- [x] pyenv shims works
- [x] clean openpilot build
- [x] factory reset
  - [x] user-prompted reset
  - [x] corrupt userdata
  - [x] finish ABL reset
- [x] color calibration
  - [x] from /persist/comma/
  - [x] directly from panel over sysfs
- [x] clean setup
- [x] cameras
- [x] update works on warm boot
  - [x] pr [... cut]
closed 2023-07-10 by adeebshihadeh