commaai/openpilot #28736: LEXUS NX-300h hybrid 2021 [vehicle is not recognized and is in dashcam mode]

Bugal1998, 2023-06-29:
### Describe the bug

First time installing comma 3 in vehicle, and installed latest openpilot firmware and vehicle isn't recognized 

### Which car does this affect?

Lexus NX 300h 2021

### Provide a route where the issue occurs

Parked in driveway 

### openpilot version

Latest

### Additional info

Running latest firmware installed but vehicle isn't recognized

sshane, 2023-06-29:
What is your dongle id? You can find it in device settings.

sshane, 2023-06-29:
Ah I see your car in our system, looks like your fingerprints were added in a recent PR to master, so you'll want to switch to master for now.

Use this installer URL on the device: https://installer.comma.ai/commaai/nightly

closed 2023-06-29 by sshane

Bugal1998, 2023-06-30:
Yup, master did the trick; everything worked perfectly on the test drive. Thanks for the help!

sshane, 2023-06-30:
Nice! Checked to make sure we're getting new ECUs I added queries for recently. Super cool, I can see what parts your hybrid drivetrain uses!

https://lexus.oempartsonline.com/oem-parts/lexus-control-module-8998078380?c=bD0xJm49U2VhcmNoIFJlc3VsdHM%3D
https://lexus.oempartsonline.com/oem-parts/lexus-computer-8998178031?c=bD0xJm49U2VhcmNoIFJlc3VsdHM%3D
https://lexus.oempartsonline.com/oem-parts/lexus-computer-8988248120?c=bD0xJm49U2VhcmNoIFJlc3VsdHM%3D

Bugal1998, 2023-06-30:
Cool.  And interesting…. You can see the ecu and other data for my vehicle even though I haven’t yet activated my prime account?

-Ken

https://lexus.oempartsonline.com/oem-parts/lexus-control-module-8998078380?c=bD0xJm49U2VhcmNoIFJlc3VsdHM%3D
https://lexus.oempartsonline.com/oem-parts/lexus-computer-8998178031?c=bD0xJm49U2VhcmNoIFJlc3VsdHM%3D
https://lexus.oempartsonline.com/oem-parts/lexus-computer-8988248120?c=bD0xJm49U2VhcmNoIFJlc3VsdHM%3D

sshane, 2023-06-30:
Yes, we have a "cars not fingerprinting" dashboard and your car popped up on it. The device always uploads its sparse and low resolution logs/videos on WiFi

Bugal1998, 2023-06-30:
Got it. It certainly made for easy probablem resolution.  Thanks again!