commaai/openpilot #28747: rawgpsd: SUPL configuration persists

adeebshihadeh, 2023-06-29:
We removed the SUPL configuration in #28278 when we switched to XTRA for AGPS, but the configuration persists, so it's unclear if the TTFF we're getting now is from SUPL, XTRA, or both. We should either add the configuration back or clear it on startup, so all the modules are in a consistent state

adeebshihadeh, 2023-07-10:
Not immediately clear how to clear this configuration. We're getting good performance now with most of our devices having both the SUPL and XTRA configured. @haraschax what do you think of just adding back the SUPL config?

haraschax, 2023-07-10:
We can add it back, though in my experiments I never got the SUPL to help at all.

adeebshihadeh, 2023-07-10:
Added back in #28858, so all devices should be in a consistent state now.

closed 2023-07-10 by adeebshihadeh