commaai/openpilot #28776: Mac: replay fails to run when ui is open

royjr, 2023-07-02:
### Describe the bug

- Was having these issues on Sonoma so started with a fresh Ventura install, installed brew, scons built successfully
- Opening ui with `./selfdrive/ui/ui` runs successfully
- Opening replay with `./tools/replay/replay --demo` either fails with the following..
```
loading route  "4cf7a6ad03080c90|2021-09-29--13-46-36"
load route 4cf7a6ad03080c90|2021-09-29--13-46-36 with 11 valid segments
Error opening terminal: xterm-256color.
QNetworkInterface: could not create IPv6 socket (Too many open files)
QNetworkInterface: could not create IPv6 socket (Too many open files)
QNetworkInterface: could not create IPv6 socket (Too many open files)
```
- ..or this after loading part of the replay ui
```
__ Engaged __ Disengaged __ Info __ Warning __ Critical __ User Tag 
QThreadPipe: Unable to create pipe: Too many open files
QEventDispatcherUNIXPrivate(): Cannot continue without a thread pipe
zsh: abort      ./tools/replay/replay --demo
```
- Closing ui and opening replay by itself runs successfully
- Opening ui while replay is running causes replay to fail with the following..
```
QNetworkInterface: could not create IPv6 socket (Too many open files)
QNetworkInterface: could not create IPv6 socket (Too many open files)
```

### OS Version

MacOS Ventura 13.4.1 (22F82) on M1 Macbook Pro

### openpilot version or commit

6495a18cc9e8c1394ac7395d24bb0a661ce35fbf

deanlee, 2023-07-03:
it is not a op bug.your system has reached the ulimit.you can fix it by googling "Too many open files"

royjr, 2023-07-03:
Interesting..

ulimit -n 256 -> `Error opening terminal: xterm-256color.`
ulimit -n 300 -> `degAssertion failed: (*fd >= 0), function malloc_with_fd, file visionbuf_cl.cc, line 24.`
ulimit -n 400 -> `degzsh: segmentation fault`
ulimit -n 999999 -> `degzsh: segmentation fault`
ulimit all unlimited -> `degzsh: segmentation fault`
[image]

default macOS values..
[image]

default ubuntu 20.04 values..
[image]

royjr, 2023-07-04:
Fixed by https://github.com/commaai/openpilot/pull/28779

closed 2023-07-04 by royjr

royjr, 2023-07-08:
I had to additionally change file descriptors from 256 to 512 by `ulimit -n 512`