commaai/openpilot #28777: Check for update bug

MJR1989, 2023-07-02:
### Describe the bug

Upon plugging in my comma 3 it automatically checks for update before wifi is connected and gives me the error “Immediately connect to the internet to check for updates. If you do not connect to the internet, openpilot won't engage in 1 day.” even though I just updated to 0.9.3 yesterday….device does not do this on old versions of openpilot only when updated to 0.9.3
[image]

### Provide a route where the issue occurs

b0a7a22dea55a1cd

### openpilot version

0.9.3

SikeItsMike, 2023-07-14:
I have the same error and so do a lot of other people from what I'm reading on discord. People seem to be attributing it to an error in the master rather than forks

adeebshihadeh, 2023-07-19:
This will be fixed in the upcoming release. You can try the `nightly` branch in the meantime if you want to give it a try; you can install with `installer.comma.ai/commaai/nightly`.

adeebshihadeh, 2023-07-19:
This should be fixed in the upcoming release. If you want to give it a try in the meantime, you can run the `nightly` branch with this URL: `installer.comma.ai/commaai/nightly`.

closed 2023-07-19 by adeebshihadeh